OllisGit/OctoPrint-PrintJobHistory #86: Not saving finished print jobs

samiam1234, 2020-09-18:
In some way I can't get the PJH plugin to work properly. I've had problems with the older PrintHistory plugin as well. I don't know if the problems are related. I've been trying hard and reading as much on the forum but I can't solve the problem. Maybe someone here can help me out.

After finishing a print I get the error "ERROR:PJH-DatabaseManager Could not insert the printjob into the database. See OctoPrint.log for details!".
This does not happen if I cancel a print, in that case the cancelled job is saved without problem.

I've tried to look at the log file. My guess is that it could have something to do with the temperature settings.
My printer setup is a bit different than most. I'm not using the temperature controlls in g-code, I'm using external temperature controllers instead. Maybe that's why it's crashing?

2020-09-18 09:31:27,469 - octoprint.plugins.PrintJobHistory.DatabaseManager - ERROR - Could not insert printJob into database:NOT NULL constraint failed: pjh_temperaturemodel.sensorValue
Traceback (most recent call last):

[octoprint (8).log](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/5244147/octoprint.8.log)

Any help is much appreciated! 
Sam

florianbeisel, 2020-09-18:
I can confirm said bug, with OctoPrint 1.4.2 Python 2.7.16 OctoPi 0.17.0 and the latest plugin version from the repository. Other than you I have a printer with "normal" GCODE controlled temperatures. 

As you mentioned I too can save a failed print job. As far as I can see this happens whether "Take temperature values from PreHeat-Plugin" is enabled or not.

florianbeisel, 2020-09-19:
I know this seems strange - and maybe it was a fluke, but playing around with this I noticed that if I delete the JobPrintHistory database and if the first print is successful it gets correctly stored in the database. After that it does not seem to be working. 

As mentioned earlier, failed PrintJobs are always added just fine. 

As I have been playing around with the settings in my attempt to get this working I might have changed something else in the configuration. I will double check whether I can reproduce this once my current print finishes.

OllisGit, 2020-09-19:
Hi,

new version 1.9.0 is out and fix the issue if the bed temperature is not available.

Please test and give feedback.

Thx in advance
Olli

somehibs, 2020-09-20:
Oh, that's why it's not recording, that explains where the last 12 hours of prints went. I'll let you know if it's working in a few hours.

somehibs, 2020-09-20:
It works now.

samiam1234, 2020-09-22:
Hi Olli,
I confirm it's working for me as well! Many thank!

closed 2020-09-22 by samiam1234