OllisGit/OctoPrint-PrintJobHistory #87: Wrong statistics

spanzetta, 2020-09-20:
See screenshots and attached CSV..
The total **Length** and **Weight** are definitively wrong
Pls check it
Thanks

[OctoprintPrintJobHistory.zip](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/5252174/OctoprintPrintJobHistory.zip)

[image]

spanzetta, 2020-09-22:
Hi @OllisGit 
any comments?
Thanks

OllisGit, 2020-09-22:
Hi @spanzetta ,

can't see any issue:
[image]

Next time, please add more informations. "definitively wrong" is not helpful!

Something like this:
```
expected value: 123, plugin-value: 321
```
Thx, in advance
Olli

spanzetta, 2020-09-23:
dear @OllisGit 
Sorry but I was sure it was so evident..
You can read in the first row of the screenshots that the Timeframe selection is ALL (Timeframe All/Filter ALL) .. but the data (First/Last) are from 01.09 to 07.09.. so as Total statistics .. data is wrong  (I wanted to calculate the Total prints)
I hope is more clear now?
If not pls let me know
Thanks

spanzetta, 2020-09-28:
Do you confirm is it clear now?
Thanks

OllisGit, 2020-10-08:
Hi @spanzetta ,

it is fixed in the latest version 1.10.0.

Please test and give feedback.

Thx, ,in advance
Olli

spanzetta, 2020-10-08:
Great news!
Thanks you
I'll test and let you know
Have a nice day

spanzetta, 2020-10-18:
Now it is fine..
Thank you so much

closed 2020-10-18 by spanzetta