OllisGit/OctoPrint-PrintJobHistory #88: Documentation improvement

florianbeisel, 2020-10-01:
[Slicer Settings Documentation](https://github.com/OllisGit/OctoPrint-PrintJobHistory/wiki/Slicer-Settings)  links to a GCODE template for Cura. But for versions prior 4.7.1 there exists a bug where having an equal sign will cause the Cura configuration to become corrupt. 

It would be smart to note that in the documentation to prevent unnecessary losses of the configuration (evenmore so because cura seems to have no fix for configuration anomalities other than delete everything)

See [Cura 4.7.1 Release Notes](https://github.com/Ultimaker/Cura/releases/tag/4.7.1)

OllisGit, 2020-10-02:
Hi @florianbeisel,
thanks for reporting.

I am not sure what you mean, because I could not reproduce the issue:

- I installed CURA 4.7..1
- Added the linked gcode settings (https://gist.githubusercontent.com/tjjfvi/75210b2ed20ed194d6eab48bf70c4f12/raw/52c5ad77b0b71d9b9770e3b3cc260e738aada157/CuraSettingsInjector.gcode) to the Start-Section of my printer
- Sliced the model and save the file (without an error)
--> The file includes values from the template

Do you have a bug-report or something else?

OllisGit, 2021-04-11:
No feedback -> closed!

closed 2021-04-11 by OllisGit