OllisGit/OctoPrint-PrintJobHistory #89: [Feature Request] CostEstimation Plugin

wilkinsalex, 2020-10-03:
Specific to your roadmap item: https://github.com/OllisGit/OctoPrint-PrintJobHistory/projects/1#card-27067931

This sounds great, it would be awesome to see the cost-estimation flow through into the print history, you ask for so much great information in the cost estimation plugin, if the print takes 2hours, and between my electricity and maintenance costs, it would be $2 worth of stuff + the filament cost, it would be awesome to have the print history plugin to take that extra $2 into effect.

IDK if this is the right place to put this comment or reference, sorry if it isn't!

OllisGit, 2021-11-13:
Hi @wilkinsalex, 
new release [1.15.0](https://github.com/OllisGit/OctoPrint-PrintJobHistory/releases/tag/1.15.0) is available and the data of the cost-plugin is now stored in the printjob.

Please test and give me a feedback. 
Thx, in advance!
Olli

wilkinsalex, 2021-12-06:
Looks like it worked great @OllisGit! thanks for the awesome upgrade!

closed 2022-01-16 by OllisGit