OllisGit/OctoPrint-PrintJobHistory #92: OSError Invalid cross-device link when using symlinked OctoPrint directory

jantman, 2020-10-18:
I'm running OctoPrint 1.4.2 on OctoPi 0.17.0, with PrintJobHistory 1.10.0.

Because of insufficient free space on my Pi's SD card, I moved the OctoPrint directory (``/home/pi/.octoprint``) to an external disk and symlinked it to its original location.

After doing this, image uploads fail with an OSError:

```
2020-10-18 12:50:43,710 - octoprint - ERROR - Exception on /plugin/PrintJobHistory/upload/snapshot/20201016-212821.jpg [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/api/PrintJobHistoryAPI.py", line 314, in post_snapshot
    os.rename(sourceLocation, targetLocation)
OSError: [Errno 18] Invalid cross-device link
2020-10-18 12:50:43,727 - tornado.access - ERROR - 500 POST /plugin/PrintJobHistory/upload/snapshot/20201016-212821.jpg (::ffff:192.168.0.24) 1475.77ms
```

The problem appears to be that this plugin uses ``os.rename`` which only works within the boundary of a filesystem, whereas ``shutil.move`` does (effectively) the same thing but also works across filesystem boundaries. I found a similar issue upstream in OctoPrint itself, https://github.com/OctoPrint/OctoPrint/issues/710 , which was fixed by using ``shutil.move`` instead of ``os.rename`` ( https://github.com/OctoPrint/OctoPrint/pull/717 ).

otm77, 2020-10-21:
I have the same 500 POST error when trying to upload an image. I run octoprint within docker. Not using symlinks. Drive has 36Gb free. Screenshots from the webcam save fine, just uploading doesn't work.

```2020-10-21 05:37:25,217 - tornado.application - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/tornado/http1connection.py", line 238, in _read_message
    delegate.finish()
  File "/usr/local/lib/python2.7/site-packages/tornado/httpserver.py", line 314, in finish
    self.delegate.finish()
  File "/usr/local/lib/python2.7/site-packages/tornado/routing.py", line 251, in finish
    self.delegate.finish()
  File "/usr/local/lib/python2.7/site-packages/tornado/web.py", line 2097, in finish
    self.execute()
  File "/usr/local/lib/python2.7/site-packages/tornado/web.py", line 2117, in execute
    **self.handler_kwargs)
  File "/usr/local/lib/python2.7/site-packages/tornado/web.py", line 192, in __init__
    self.initialize(**kwargs)
  File "/data/plugins/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 3786, in initialize
    super(OctolapseLargeResponseHandler, self).initialize(
TypeError: super(type, obj): obj must be an instance or subtype of type
2020-10-21 05:43:41,896 - tornado.access - ERROR - 500 POST /plugin/PrintJobHistory/upload/snapshot/20201021-001800.jpg (192.168.0.10) 110.90ms```

jantman, 2020-10-21:
@otm77 Just for clarity, that's not the same error; the 500 is just how the web server responds. The actual error is the second-to-last line, which are quite different. Even more importantly, though, it appears that error is coming from the OctoLapse plugin, not PrintJobHistory. You should probably open an issue on https://github.com/FormerLurker/Octolapse

OllisGit, 2021-01-13:
Hi @jantman,

thanks for the bug report and also the hint for the fix.
I will implement that in one of the next releases.

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

jantman, 2021-01-13:
Thanks!!

No worries at all, we all need a sabatical from our projects now and then. Hope you got to enjoy some relaxation!

OllisGit, 2021-04-11:
fyi: Currently I have some "private challenges to solve", thats the reason why It takes so long to release a new version, sorry!

Today I released a new development version (1.11.0). This issue is now fixed. Please test and give feedback.

Thx, in advance for testing
Olli

closed 2021-05-16 by OllisGit