OllisGit/OctoPrint-PrintJobHistory #95: Save function not being found

MadeforMaking, 2020-11-03:
I've not been able to make changes to my print-jobs recently due to the save function not being successful. The developers console shows a 404 for this PUT request:

"PUT http://octopi.local:5001/plugin/PrintJobHistory/storePrintJob/66 404 (NOT FOUND)"

Any idea what could be happening or what I should try and do.

I'm running 2 instances on 1 pi, would this be an issue for the plugin?

OllisGit, 2021-01-13:
Hi @MadeforMaking,

please look into the Storage file location of the plugin. If both instances use the same database file location it won't work.
[image]

MadeforMaking, 2021-01-17:
I will have a look this week and will get back to this issue, thanks.

MadeforMaking, 2021-01-18:
The 2 paths are different
`/home/pi/.octoprint/data/PrintJobHistory/printJobHistory.db`
`/home/pi/.octoprint2/data/PrintJobHistory/printJobHistory.db`

I haven't been near octoprint in a while, so something must have changed after restarting the system. I am able to save changes so I am closing this issue.

closed 2021-01-18 by MadeforMaking