OllisGit/OctoPrint-PrintJobHistory #97: Request for recording temperature plots

Sembazuru, 2020-11-11:
The current temperature recording doesn't work for me because I use an anti-ooze nozzle temperature for mesh bed leveling so all my prints are showing a tool temperature of 160°C. Even if I didn't this would only capture the first layer print temperature, not the bulk of the print temperature. What would be more useful for me, especially for troubleshooting temperature related issues when I'm not around the printer, would be to capture the temperature plot. I realize that this may be a large dataset, especially for a multi-day print. Might need some sort of compression for what gets stored in the database.

closed 2020-12-21 by stale[bot]

OllisGit, 2021-01-13:
Hi @Sembazuru,
I was also thinking about collection more data depending on a print (print time per layer, fila-length per layer, etc.)
All these data should be stored in a timeseries-database (https://en.wikipedia.org/wiki/Time_series_database).

But currently this feature is not on my roadmap for the next month.

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

Sembazuru, 2021-01-14:
Hey, there. I was concerned with both issues being auto-closed for going stale. I hope you feel refreshed after your sabbatical. :-)

Even though the extra data collection isn't on your short-term road map, I'm encouraged that you have also been thinking about expanding the data collection.

newmanjosh, 2021-04-15:
Is there a workaround for this issue? I do the same thing to prevent oozing.  Is there maybe a way to trick it into using the correct temperature by changing the starting gcode?

Sembazuru, 2021-04-15:
I'd have to play with it. If my suspicion is correct maybe a `M104 S[temperature]` right before the anti-ooze temperature M104 in the startup gcode might be enough to fool the plugin in its current state to capture the temperature for the bulk of the print. (PrusaPrinter style variable substitution in my example. Adjust for your slicer if you don't use PrusaSlicer.)