OllisGit/OctoPrint-PrintJobHistory #99: [Feature request] Show edit dialog only on failed prints

MadeforMaking, 2020-11-14:
If I have a good print setup and things are working well with the data gathered, I don't really want to see the dialog after every print. I would like edit the job for when there is a problem and the print has failed.

It encourages me to enter information when there is a problem. This way I can look at the data and see trends etc and see my explanation.

OllisGit, 2021-01-13:
Hi @MadeforMaking ,

good idea. I will implement this enhancement in one of the next releases.

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

MadeforMaking, 2021-01-17:
Thanks @OllisGit and we all need time away. Your doing great work.

OllisGit, 2021-04-11:
fyi: Currently I have some "private challenges to solve", thats the reason why It takes so long to release a new version, sorry!

Hi @MadeforMaking, 
today I released a new development version (1.11.0). This feature is now available, please test and give feedback.

Thx, in advance for testing
Olli

closed 2021-05-16 by OllisGit