OllisGit/OctoPrint-PrintJobHistory #100: Simplify 3d

gdcircuit, 2020-11-17:
Please add support for capturing simplify 3D Slicer settings which follow a different format.

gdcircuit, 2021-02-06:
> Hi @gdcircuit <https://github.com/gdcircuit>  ,
>
> please provide a sample file.
>
> PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

Here is an example simplify 3d file.

OllisGit, 2021-04-11:
Hi @gdcircuit,

issue-answering with attaching an file is not possible with GitHub, you need to use the Web-UI to upload attachments.
BR
Olli

RobinAHoward, 2021-04-16:
Does this work?

[flowRateCalibrationCube_0.8mmWalls.txt](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/6327382/flowRateCalibrationCube_0.8mmWalls.txt)

gdcircuit, 2021-05-01:
Sorry for the late response. The email got buried. Yes this looks great. When will it be in a release? I am looking to upgrade to a new release because my plugin hasn’t been working for over a month. I have seen others with a similar issue so I figured a new release would fix it.

OllisGit, 2021-05-16:
Hi,

new version 1.3.0 is out and now the Simplify 3d - Settings could be parsed. The current solution is not the best, but should work (see #144)

Please test and give feedback.

Thx, in advance
Olli

RobinAHoward, 2021-06-11:
Looking good.  Tried it on 4 prints and doing what it should.  THANK YOU!

closed 2021-10-30 by OllisGit