OllisGit/OctoPrint-PrintJobHistory #103: [Feature Req] Easily compare details of multiple selected jobs

peta, 2020-12-09:
Everytime I try to dial-in the ideal params/values for a new filament or after changing something in the printer setup, I print many (many!) models where (few) values differ. Most of the time I just mark the printed object with a permanent marker with an incrementing number in order to keep them apart and keep track of what value I did change on which element from that series.  Thanks to the slice settings plugin support of PJH I can dive into the raw settings values (embedded as comment in the Gcode from e.g. Cura) and find out even the next day what a specific slicer setting was. But in general I find this approach rather cumbersome, since it needs many clicks and then a CTRL+F in order to locate the specific settings by its machine readable name.

## User story
As a PJH user I want to be able to select multiple print jobs and compare their slicer settings in a intuitive and easy to use way. Such a comparison view should be designed in a way that allows quick visual navigation (through the myriads of slicer settings) so that differences and specific values can be identified quickly.

## UI Mock

### Where to integrate?

[image]

### Comparison view

Just some plain tabular Excel mockup; I hope you get the point. Such a view could be viewed in a modal overlay dialog and be horizontally scrollable depending on the number of jobs selected for comparison. Lines with differences shall be highlighted.
Displaying just the machine readable settings names (Cura-flavor in my demo case) might be sufficient, but resembling the settings pane layout as well as the descriptive namings from e.g. Cura would be very awesome, since (not-so-senior-professional) users can compare/identify the settings more easily since they already know it (along with its position/layout) from Cura GUI.

[image]

## Discussion

I am aware that this is not a piece of cake. :D But in general, does somebody else see a major benefit in such a feature? How would (already do?) you accomplish such a settings series comparison demand as of today?

OllisGit, 2021-01-17:
Hi @peta,

thanks for the very detailed request-description (that's how I like it).

This is a really good idea and brings a lot of benefit to everybody.
I think an simple version of this feature could implemented easily. Additional features like "show only diffs", "grouping" or "filter/search keys" could be implemented later.

Currently I am not sure if I should put this this to my short- or mid-term roadmap.

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

OllisGit, 2021-01-27:
fyi: just a small update. implementation is almost done...only some basic features, like show all keys and colorize the diff result.
So, there are a lot of improvements:
- Better table-coloring (e.g. key column with different background)
- Value-Filter (only diff, only not present, ...)
- Key-Filter (as you type, remove other keys)
- Expand Dialog via mouse
- ...
[image]

OllisGit, 2021-04-11:
fyi: Currently I have some "private challenges to solve", thats the reason why It takes so long to release a new version, sorry!

Hi @peta, 
today I release a new development version (1.11.0). A "basic implementation" is now available.
Please test and give feedback.

Thx, in advance for testing
Olli

peta, 2021-06-23:
First of all: a big and warm THANK YOU for your blazing fast (and executable  response in regard to my feature suggestion. Just awesome. 💪😎

Back then I just started to dive into OctoPrit plugin development with your PJH as starting base in order to see if and how I could setup a poc by myself. But as you noticed I became pretty quiet since then. Unfortunately I didn't have the time yet to spend any reasonable time on the 3d printing hobby and thus didn't have the chance to thoroughly test and refine your first implementation. But it's still on my todo list, and I will come back to it as soon as i can.

OllisGit, 2021-10-30:
Please raise a new ticket for a finding or feature request...if you still working with 3D printers ;-)

closed 2021-10-30 by OllisGit