OllisGit/OctoPrint-PrintJobHistory #104: Please use data already provided by the slic3r

AndKe, 2020-12-20:
PrintJob history can display details found in the job, like: 
; filament cost = 9.3
; filament used = 40.5g

but it does not use them.
There is not much point in maintaining filament collection/prices in  more than one place.
It is logical to have filaments in the slicer, or  Slic3r - because that's where per-filamnet tuning like temperature and linear-extrusion coefficients are.

OllisGit, 2021-01-17:
Hi @AndKe ,

it's a good idea...so the user should have a choice from which source the filament-data should be taken, right?

Please provide a sample file with a description which data you want to grab from the g-code. 
Maybe you have also an idea how I could design the plugin-settings for this feature.

BR
Olli
PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

AndKe, 2021-01-17:
I would suggest to have a simple checkbox in the settings:
"use slicer-provided data when available"
please see the great goodies you'll get right in the .gcode:

```
; **generated by Slic3r** 1.3.1-dev (Build 54a31ee) on 2021-01-04 at 20:53:09

; Print Config Notes: 
; reduced infill speed from 80 to 60 as a test
; 
; Filament notes: 
; Kapton+ABS juice : good adhesion with raft  at 93deg
; UltraBase: some warping at 110deg 
; BuildTec: VERY good bed adhesion at 85°C   , 70°C
; 
; 250°C extrusion  good layer adhesion, 238 is too low.
; 
; Printer Config Notes: 
; 4mm retraction is fine. 
; 5mm jammed with white PLA  -
; 
; E0 is a bit too strong
; 
; Linear Advance: http://marlinfw.org/tools/lin_advance/k-factor.html
; M900 K0.6 ;  ok for PLA at 210°C
; M900  K4 ; ok for TPU , calibration shows that  a value of 2-2.4 is good, should be tested
; 
; BuildTak 
; 
; 1. PLA:50-70℃
; 2. Flexible filament: 50-70℃
; 3. ABS:100-125℃ ( 110℃ is recommended )
; 4. PC:100-130℃ ( 120℃ is recommended )
; 5. Nylon:90-120℃ ( 110℃ is recommended )
; 6. PP:100-130℃ ( 120℃ is recommended )
; 7. PETG: 50-70℃
; 
; ABS 80° , @250° - still warping
; 
; 
; AnyCubic Ultrabase: (Glass)
; PLA: 50°C is good.
; TPU : 40°C is good.
; 
; CCTREE magnetic Surface;
; ABS 80° , @250° - still warping
; PLA: 50  seems plenty
; TPU : 70 is too sticky 
; 
; PEI-double-sided steel, no-name;
; ABS: POOR adhesion at 80° +250°
; PLA: POOR  adhesion at 55-65°C
; TPU: decent adhesion already at 40°C
; 
; external perimeters extrusion width = 0.44mm (1.59mm^3/s)
; perimeters extrusion width = 0.56mm (4.14mm^3/s)
; infill extrusion width = 0.50mm (5.48mm^3/s)
; solid infill extrusion width = 0.56mm (2.07mm^3/s)
; top infill extrusion width = 0.56mm (1.55mm^3/s)

G92 E0
M190 S77 ; set bed temperature and wait for it to be reached
M104 S260 ; set temperature
M851 Z-1.50  ;Z Probe Offset in firmware =0   // negative means more squished  
;M85 S720 ; kill job if no activity for 12min
G28 ; home all axes
;G29 ;level bed
G1 X300 F8000 ;park at end of X axis , present print , feedrate
G1 X318 F700 ;park X axis , feedrate
; M85 S300 ; kill job if no activity for 5min
[... 187 lines cut ...]
; infill_every_layers = 1
; infill_extruder = 1
; infill_extrusion_width = 0
; infill_overlap = 55%
; infill_speed = 60
; min_shell_thickness = 0
; min_top_bottom_shell_thickness = 0
; overhangs = 1
; perimeter_extruder = 1
; perimeter_extrusion_width = 0
; perimeter_speed = 40
; perimeters = 3
; small_perimeter_speed = 15
; solid_infill_below_area = 70
; solid_infill_every_layers = 0
; solid_infill_extruder = 1
; solid_infill_extrusion_width = 0
; solid_infill_speed = 20
; thin_walls = 1
; top_infill_extrusion_width = 0
; top_infill_pattern = rectilinear
; top_solid_infill_speed = 15
; top_solid_layers = 2

```

AndKe, 2021-01-19:
PrusaSlicer  (based on Slic3r - and in active development) has also
```
; filament_colour = #29B2B2     ################## HERE IS THE COLOR
; filament_cooling_final_speed = 3.4
; filament_cooling_initial_speed = 2.2
; filament_cooling_moves = 4
; filament_cost = 0
; filament_density = 0
; filament_diameter = 1.75
; filament_load_time = 0
; filament_loading_speed = 28
; filament_loading_speed_start = 3
; filament_max_volumetric_speed = 0
; filament_minimal_purge_on_wipe_tower = 15
; filament_notes = ""      ################  HERE could the manufacturer be...
; filament_ramming_parameters = "120 100 6.6 6.8 7.2 7.6 7.9 8.2 8.7 9.4 9.9 10.0| 0.05 6.6 0.45 6.8 0.95 7.8 1.45 8.3 1.95 9.7 2.45 10 2.95 7.6 3.45 7.6 3.95 7.6 4.45 7.6 4.95 7.6"
; filament_settings_id = PLA with a name  ################## HERE IS THE NAME
```

OllisGit, 2021-04-11:
Added some implementation notes for myself:

- Plugin-Settings: for each attribute it needs a mapping. E.g. Diameter = filament_diameter
- Before a print starts and a "filament-selection" was detected in gcode, show the filament-attributes to the user. So that he/she is able to cancel the job
- OPEN: Handling "multi-tool-support"

@AndKe do you have a sample which  use multi-extruders? Is there an added number behind the keys, like  ```filament_diameter3```?

AndKe, 2021-04-11:
@OllisGit 
A quick test shows me that repeating output is for next item:
This is output from a two-extruder job:

; filament used = 761.8mm (1.8cm3)
; filament used = 2.2g
; filament cost = 0.5
; filament used = 73.9mm (0.2cm3)
; filament used = 0.2g
; filament cost = 0.0
; total filament cost = 0.6

OllisGit, 2022-01-17:
fyi: I am starting writing a kind of concept to implement such feature. I am looking for a simple and easy to use solution.
https://github.com/OllisGit/OctoPrint-PrintJobHistory/wiki/GCodeSpoolInfo