OllisGit/OctoPrint-PrintJobHistory #105: No statistic saved

moddok, 2020-12-21:
Hi,

the statistics don't work for me. It's just empty.
It's an standard installation without any customization.
I have installed PreHeat (0.5.1), FilamentManager (1.6.3) and DisplayLayerProgress (1.24.0) too.
OctoPrint: 1.5.2
PrintJobHistory: 1.10.0

Any tips on how to troubleshoot and/or fixing it?

Thank you!!

moddok, 2020-12-21:
Edit2: Found some errors in the octoprint log

`2020-12-21 10:49:54,692 - octoprint - ERROR - Exception on /plugin/PrintJobHistory/loadStatisticByQuery [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/api/PrintJobHistoryAPI.py", line 177, in get_statisticByQuery
    statistic = self._databaseManager.calculatePrintJobsStatisticByQuery(tableQuery)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/DatabaseManager.py", line 418, in calculatePrintJobsStatisticByQuery
    if (StringUtils.isEmpty(filla.spoolName) == False):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/common/StringUtils.py", line 192, in isEmpty
    if (value == None or len(str(value).strip())==0 ):
UnicodeEncodeError: 'ascii' codec can't encode character u'\xfc' in position 8: ordinal not in range(128)
2020-12-21 10:49:54,706 - tornado.access - ERROR - 500 GET /plugin/PrintJobHistory/loadStatisticByQuery?from=0&to=25&sortColumn=printStartDateTime&sortOrder=desc&filterName=all&startDate=&endDate= (::ffff:192.168.1.8) 149.35ms
`

But i don't know what to do now :D

moddok, 2020-12-21:
Found it!

Got an **Ü** in the name of a spool. Renaming the role and deleting the old prints worked.

OllisGit, 2021-01-17:
Hi @moddok,
sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

I will take a look an this in the next days.

OllisGit, 2021-04-11:
fyi: Currently I have some "private challenges to solve", thats the reason why It takes so long to release a new version, sorry!

Hi @moddok, 
today I release a new development version (1.11.0). This fixes the issue. Please test and give feedback

Thx, in advance for testing
Olli

closed 2021-05-16 by OllisGit