OllisGit/OctoPrint-PrintJobHistory #108: CVS Import append not working.

BMendelson, 2021-01-05:
I have a CVS file from a previous history plugin that I am trying to append to this database. I've reformatted the CVS file so it meets the import criteria. I get a bunch of database errors. Log shows error reading line * in  column ' '. Attached log file, cvs file and screenshot of errors.
[PrintHistory bug.zip](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/5771365/PrintHistory.bug.zip)

OllisGit, 2021-01-17:
Hi @BMendelson !

Wow, I should definitely improve the log-error output popup (#112)!

Thx, for the provided data I will look to this in the next days.

OllisGit, 2021-04-11:
fyi: Currently I have some "private challenges to solve", thats the reason why It takes so long to release a new version, sorry!

Today I released a new development version (1.11.0). This fixed the issue, please test and give feedback.

Thx, in advance for testing
Olli

closed 2021-05-16 by OllisGit