OllisGit/OctoPrint-PrintJobHistory #109: used length / weight alway 0 whith MK3s + MMU

diedder, 2021-01-07:
every job is finished with 0g used filament in the database. I'm using a MMU2S with the octoprint mmu filament plugin. There is no difference which tool in software or filament port I am using, even in single filament mode this issue occurs.

OllisGit, 2021-01-17:
Hi @diedder ,

please change the plugin log-level to debug, do a print and attach the octoprint.log to this issue.
Please also check with defaault-tool do you selected in the plugin-settings. Currently the plugin only supports a single extruder.
[image]

[image]

diedder, 2021-01-30:
Hi @OllisGit 
the used MMU filament select plugin isn't working anymore. See also my comment at the filament manager. I think without this plugin ocotprint doesn't know which tool is used?

diedder, 2021-02-16:
Hi @OllisGit 
since the last update the plugin stopped working with my setup (MMU and MMU in single filament mode). The were no entries foo jobs added in the db. The popup after a job is not showing up.
I activated the logging as requeted, which file do you need? the octopint.log is 34MB?

OllisGit, 2021-04-11:
Hi @diedder,

Today I released a new development version (1.11.0). Maybe this fix this issue. Please test and give feedback.

Thx, in advance
Olli

diedder, 2021-04-18:
Hi @OllisGit ,

thanks for picking up this issue, today I tried two test-prints with PrintJobHistory (1.12.dev1) an new installed SpoolManager Plugin (1.3.4). The SpoolManager also didn't worked with my MMU2S setup.

After printing in SingleMaterialMode (multi filament not tested yet) the job popup after finishing doesn't appear. No new job is displayed in the list. After rebooting the server for another reasons the job is visible in the list...

The SpoolManager ist also substracting filament of one test-spool!

So half way gone? Do you need further informations?

OllisGit, 2021-05-24:
Try to use the lastest PJH-Plugin V1.14.0 and SpoolManger V1.4.0, that should work!

Please test and give feedback!
Olli

diedder, 2021-05-25:
I tried with 1.4.2 in two modes.

first of all, the popup after finishing a job appears!

sliced for MK3S MMU2 machine in single filament mode:
I'm using a plugin to select the tool in single mode via octoprint, instead of this the choice can be made on the printer.
The popup befode start showed tool0. After that I selected tool3 via the plugin, after finishing the job the calculation was on tool0, withoug weight, only length
[image]
but at the sidebar of the spoolmanager the correct weight ob tool3 was substracted (was 729,8g before, job was 8.3g)
[image]

second try - sliced with multi tool, selected tool3 for this print in slicer, so it should be in the gcode:
the popup before start asked for tool3 - correct. after the print the calculated weight was shown correct.
[image]
in the sidebar the correct tool3 was substracted
[image]

So there seems to be an issue with the single filament mode. As you can see in the last screenshot, the parallel used filament plugin substraced in both tries the correct tool (alls tools were 1000g before the test), so it should be possible to fix this.

If you need more tests or data pls let me know.

A big "thank you" for implementing the multi tool support! 

diedder

OllisGit, 2021-05-25:
Hi,

please provide me:
- the two gcode files
- the name of the plugin you use
- each spools has values for diameter and density, right?
- octoprint.log

Thanks goes to @codingcatgirl, because she raised a PullRequest for that feature.

codingcatgirl, 2021-05-25:
You don't have to thank me every time, but you're welcome <3

diedder, 2021-05-26:
> Hi,
> 
> please provide me:
> 
> * the two gcode files
> * the name of the plugin you use
> * each spools has values for diameter and density, right?
> * octoprint.log
> 
> Thanks goes to @codingcatgirl, because she raised a PullRequest for that feature.

Hi,

unfortunately I deleted one file... I've uploaded the remaning file an additionally sliced a single cubus in the both described ways, MMU single mode and MMU tool2

[Halterahmen_PETG@MMU_r0.3x0.4-s2-i15-n240-b90_MK3S+MMU2S__27m.gcode.txt](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/6547945/Halterahmen_PETG%40MMU_r0.3x0.4-s2-i15-n240-b90_MK3S%2BMMU2S__27m.gcode.txt)
[MMU-Single_PETG_r0.3x0.4-s2-i20-n240-b90_MK3S+MMU2S-Single__33m.gcode.txt](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/6547947/MMU-Single_PETG_r0.3x0.4-s2-i20-n240-b90_MK3S%2BMMU2S-Single__33m.gcode.txt)
[MMU-tool3_Generic PETG @MMU2_r0.3x0.4-s2-i20-n230-b90_MK3S+MMU2S__33m.gcode.txt](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/6547948/MMU-tool3_Generic.PETG.%40MMU2_r0.3x0.4-s2-i20-n230-b90_MK3S%2BMMU2S__33m.gcode.txt)

All spools have density an diameter. The plugin to select tool in single mode is this one
https://github.com/tkoecker/OctoPrint-Mmu2filamentselect

my octoprint.log is > 10MB and very long... Are there any settings to clean up and focus on what you want to ivnestigate?

closed 2022-02-28 by github-actions[bot]