OllisGit/OctoPrint-PrintJobHistory #114: No Job History being saved

rmknightz, 2021-01-31:
Hi,

I have Updated my Octopi last week and I know realize I am not getting the option to save prints.

This has been working flawlessly for the last couple of months.

From the attached log file I get

021-01-31 21:17:47,889 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Try taking snapshot '/home/pi/.octoprint/data/PrintJobHistory/snapshots/20210131-204139.jpg' from 'http://127.0.0.1:8080/?action=snapshot'
2021-01-31 21:17:47,895 - octoprint.plugin - ERROR - Error while calling plugin PrintJobHistory
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 614, in on_event
    self._printJobFinished("success", payload)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 436, in _printJobFinished
    self._createAndAssignFilamentModel(self._currentPrintJobModel, payload)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 254, in _createAndAssignFilamentModel
    filemanentModel.usedLength = self._filamentManagerPluginImplementation.filamentOdometer.totalExtrusion[0]
AttributeError: 'FilamentManagerPlugin' object has no attribute 'filamentOdometer'
2021-01-31 21:17:47,949 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Process snapshot image
2021-01-31 21:17:47,993 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Image stored to '/home/pi/.octoprint/data/PrintJobHistory/snapshots/20210131-204139.jpg'
2021-01-31 21:17:48,019 - octoprint.plugins.filamentmanager - INFO - Updating Filament usage for octoprint configured toolcount: 1
2021-01-31 21:17:48,019 - octoprint.plugins.filamentmanager - INFO - Filament tracked toolcount: 1
2021-01-31 21:17:48,020 - octoprint.plugins.filamentmanager - INFO - Filament tracked values: [1378.4003899999984]
2021-01-31 21:17:48,021 - octoprint.plugins.filamentmanager - INFO - Filament used: 1378.40039 mm (tool0)
2021-01-31 21:17:48,054 - octoprint.plugins.tracking - INFO - Sent tracking event print_done, payload: {u'origin': u'local', u'file': '7ea1581e8e97f8e8f3a03bfb56adb5becf444193', u'elapsed': 2166}
2021-01-31 21:17:48,082 - octoprint.plugins.filamentmanager - INFO - Updated remaining filament on spool 'Silky Copper - PLA (MKOEM)' from 861.372g to 857.227698156g (-4.1443018439g)
2021-01-31 21:17:48,107 - octoprint.plugins.telegram - ERROR - An Exception in get final time : unsupported type for timedelta seconds component: unicode
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_telegram/__init__.py", line 1465, in calculate_ETA
    finish_time = current_time + datetime.timedelta(0,printTime)
TypeError: unsupported type for timedelta seconds component: unicode

My current system details.
browser.user_agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 11_1_0) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/88.0.4324.96 Safari/537.36
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octopi2.knightz.co.uk
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 3959984128
env.os.bits: 32
env.os.id: linux
env.os.platform: linux2
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.17.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.3.3
env.python.version: 2.7.16
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.3
printer.firmware: Marlin bugfix-2.0.x (Dec 16 2020

[octoprint (1).log](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/5900665/octoprint.1.log)

OllisGit, 2021-01-31:
Hi @rmknightz,

I think you didn't read the release notes of the last version of FilamentManager.
https://github.com/OllisGit/OctoPrint-FilamentManager/releases 

Wait or go back to Version 1.7.0.

rmknightz, 2021-02-01:
@OllisGit 

My bad. Thanks for the swift reply.

apascualm, 2021-02-12:
> Hi @rmknightz,
> 
> I think you didn't read the release notes of the last version of FilamentManager.
> https://github.com/OllisGit/OctoPrint-FilamentManager/releases
> 
> Wait or go back to Version 1.7.0.

How I can to make a rollback to 1.7.0 version??

Thanks

jolcese, 2021-02-12:
> > Hi @rmknightz,
> > I think you didn't read the release notes of the last version of FilamentManager.
> > https://github.com/OllisGit/OctoPrint-FilamentManager/releases
> > Wait or go back to Version 1.7.0.
> 
> How I can to make a rollback to 1.7.0 version??
> 
> Thanks

Install from here: https://github.com/OllisGit/OctoPrint-FilamentManager/releases/download/1.7.0/master.zip

DarkDalmatian, 2021-02-26:
I cannot even uninstall the new version... I get error!!!!

OllisGit, 2021-04-11:
Hi @DarkDalmatian,

what kind of error do you receive during uninstalling the plugin?
Did you solve the uninstall issue? -> e.g. deleting "by hand"!?

OllisGit, 2021-04-11:
fyi: Currently I have some "private challenges to solve", thats the reason why It takes so long to release a new version, sorry!

Today I released a new development version (1.11.0). This should fix the issue.

Thx, in advance for testing
Olli

DarkDalmatian, 2021-04-12:
@OllisGit it's fine, I did the manual trick suggested in the other topc and it's now fine. Basically just changed a letter from lower case to bigger case or something similar...

closed 2021-04-12 by OllisGit