OllisGit/OctoPrint-PrintJobHistory #115: No history being saved after update octopi 0.18.0 octoprint 1.5.3

doktourtv, 2021-02-02:
Print history was working fine until I upgraded to octopi 0.18.0 / octoprint 1.5.3. The pop up window does not appear after prints (successful or unsuccessful).

For the attached log I started octoprint, began a print then canceled. No popup and no history was saved.

Please advise if I am doing something incorrectly. (I did toggle _'always_' for '_capture print history'_ and _'show print job edit dialog after print_" )

I am running version 1.10.0 of PrintJobHistory (latest from github)

[octoprint (5).log](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/5908515/octoprint.5.log)

samwiseg0, 2021-02-02:
I am running into the same issue. Please let me know if you need any other debug info.

samwiseg0, 2021-02-02:
@doktourtv nvm I found the answer in another thread. This one should be closed...

https://github.com/OllisGit/OctoPrint-PrintJobHistory/issues/114#issuecomment-770459976

cmoe71, 2021-02-09:
Have the same issue (OctoPrint 1.5.3, Python 3.7.3, OctoPi 0.18.0)

[octoprint.log.2021-02-08.txt](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/5954147/octoprint.log.2021-02-08.txt)

[Systeminfo.txt](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/5954126/Systeminfo.txt)

Thank you, Clemens

doktourtv, 2021-02-09:
> Have the same issue (OctoPrint 1.5.3, Python 3.7.3, OctoPi 0.18.0)
>
> [octoprint.log.2021-02-08.txt](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/5954147/octoprint.log.2021-02-08.txt) [...]

this will solve your problem:
https://github.com/OllisGit/OctoPrint-PrintJobHistory/issues/114#issuecomment-770459976

cwbullet, 2021-02-25:
Same issue.

OllisGit, 2021-04-11:
duplicated, see #118 
New version 1.11.0-dev should solve this issue.

closed 2021-04-11 by OllisGit