OllisGit/OctoPrint-PrintJobHistory #118: No Print Job History is being saved

johnbozuk, 2021-02-21:
Nothing is being saved in Print Job History when a print completes or has an error, was using latest stable release of Octoprint which was not working, uninstalled print job history and reinstalled still nothing, I have now upgraded to the release version of octoprint and still nothing.

tenpawelnowak, 2021-02-22:
Refer to release for 1.10.0 notes https://github.com/OllisGit/OctoPrint-PrintJobHistory/releases which doesn't work with FilamentManage >= V1.7.2+. Downgrade to 1.7.0 and your history will start showing. For me even pre 1.7.0 printes showed up.

johnbozuk, 2021-02-22:
Ok, I think I removed filament manager last night as hoping spool manager
is going to work with print job history soon or do I still need both
installed?

How do I downgrade anyway?

Thanks

johnbozuk, 2021-02-22:
Have now downgraded to 1.7.0 and yes they are now saving, so a problem in a later release.

Any ideas why I keep getting Insufficient filament warnings when I have loads of metres left on the spool and in the settings?

MrDragon420, 2021-02-24:
When will this be fixed? I like using the filament manger plugin

cwbullet, 2021-02-25:
I downgraded to 1.7 and it worked for a week and now non of my printers are saving the history.

DarkDalmatian, 2021-02-26:
I have also the issue, will be nice to have a quick fix for this BIG issue instead of waiting for a full new release...

cwbullet, 2021-02-26:
this is not just a filament manager problem.

locki-cz, 2021-03-03:
looking forward to fixed version, I instaled this plugin and after week of printing nothing in history, instaled uninstlaed ... nothing changed and then i found this thread 👍

tonnic, 2021-03-09:
I am using 1.7 and there is no history being saved...

MikeSiekkinen, 2021-03-09:
I was having the same issue.  Can't say if the same root cause for everyone here but I noticed in octoprint log

```2021-03-08 17:37:24,595 - octoprint.plugin - ERROR - Error while calling plugin PrintJobHistory
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 614, in on_event
    self._printJobFinished("success", payload)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 436, in _printJobFinished
    self._createAndAssignFilamentModel(self._currentPrintJobModel, payload)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 254, in _createAndAssignFilamentModel
    filemanentModel.usedLength = self._filamentManagerPluginImplementation.filamentOdometer.totalExtrusion[0]
AttributeError: 'FilamentManagerPlugin' object has no attribute 'filamentOdometer'
```

I submitted a pull request for a proposed fix.  In the mean time if you want to patch yourself it's just that one line change.  I had to manually purge related .pyc files for PrintJobHistory, restarted Octoprint for good measure.

tonnic, 2021-03-09:
> I was having the same issue. Can't say if the same root cause for everyone here but I noticed in octoprint log
> 
> ```
> Traceback (most recent call last):
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
>     result = getattr(plugin, method)(*args, **kwargs)
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
>     return f(*args, **kwargs)
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 614, in on_event
>     self._printJobFinished("success", payload)
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 436, in _printJobFinished
>     self._createAndAssignFilamentModel(self._currentPrintJobModel, payload)
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 254, in _createAndAssignFilamentModel
>     filemanentModel.usedLength = self._filamentManagerPluginImplementation.filamentOdometer.totalExtrusion[0]
> AttributeError: 'FilamentManagerPlugin' object has no attribute 'filamentOdometer'
> ```
> 
> I submitted a pull request for a proposed fix. In the mean time if you want to patch yourself it's just that one line change. I had to manually purge related .pyc files for PrintJobHistory, restarted Octoprint for good measure.

So what do we need to do?

MikeSiekkinen, 2021-03-09:
> So what do we need to do?

If you're not familiar with reading pull request source code or comfortable with manually patching then you'd need to wait for the fix to be officially merged into a release.

truglodite, 2021-03-13:
> So what do we need to do?

Seems this should get pushed out sooner than later lol... here you go:

Refer to the merged PR to find the required change (line 254, change `filamentOdometer` to `myFilamentOdometer`):
https://github.com/OllisGit/OctoPrint-PrintJobHistory/pull/122/commits/69c2f1630c78429b1df2543ae4ed3ebe81c80e72

SSH in to your octoprint, and edit the file on your instance, which should be located here on current installs:
`/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/__init__.py`

After saving the change to the `__init__.py` file, go to the cache files for printjobhistory, located here:
`/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/__pycache__`

Delete all the *.pyc files in there. Now reboot octoprint and enjoy.

tonnic, 2021-03-13:
> > So what do we need to do?
> 
> Seems this should get pushed out sooner than later lol... here you go:
> 
> Refer to the merged PR to find the required change (line 254, change `filamentOdometer` to `myFilamentOdometer`):
> [69c2f16](https://github.com/OllisGit/OctoPrint-PrintJobHistory/commit/69c2f1630c78429b1df2543ae4ed3ebe81c80e72)
> 
> SSH in to your octoprint, and edit the file on your instance, which should be located here on current installs:
> `/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/__init__.py`
> 
> After saving the change to the `__init__.py` file, go to the cache files for printjobhistory, located here:
> `/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/__pycache__`
> 
> Delete all the *.pyc files in there (they will reappear after rebooting). Now reboot octoprint and enjoy.

Awesome, you rock.  Does this still mean that I am still limited to v1.7.0?

truglodite, 2021-03-13:
> ... Does this still mean that I am still limited to v1.7.0?

No, this will fix 1.7.2.

tonnic, 2021-03-13:
> > ... Does this still mean that I am still limited to v1.7.0?
> 
> No, this will fix 1.7.2.

So I can only use up to 1.7.2 then

tonnic, 2021-03-15:
> > ... Does this still mean that I am still limited to v1.7.0?
> 
> No, this will fix 1.7.2.

This did not work for me.  My line looks like:

`filemanentModel.usedLength = self._filamentManagerPluginImplementation.myfilamentOdometer.totalExtrusion[0]
`
Still not writing to my history.
- Filament Manager (1.7.2)
- PrintJobHistory (1.10.0)

truglodite, 2021-03-15:
Use a capital F.

tonnic, 2021-03-15:
> Use a capital F.

Oh dang...didn't even see that and at one point was like "I wonder why they didn't capitalize the F"... thanks!, will try that now.

spanzetta, 2021-03-16:
same issue here... I see the comments about manual changes:
_SSH in to your octoprint, and edit the file on your instance, which should be located here on current installs:
/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/__init__.py
After saving the change to the __init__.py file, go to the cache files for printjobhistory, located here:
/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/__pycache__
Delete all the *.pyc files in there (they will reappear after rebooting). Now reboot octoprint and enjoy_

but I don't have a /home/pi/oprint/lib/**python3.7**/ but /home/pi/oprint/lib/**python2.7**
Are these instructions still valid?

I also don't see the folder  /home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/__pycache_ (missing)
Is that normal?

Thanks in advance

spanzetta, 2021-03-16:
I answer myself...
I did apply the suggested fix at /home/pi/oprint/lib/python2.7 folder.. and it works now!
Thanks

DarkDalmatian, 2021-03-16:
Same as @spanzetta !! 
folder /home/pi/oprint/lib/python2.7 and no cache folder, changed the file, restarted Raspberry Pi,printed... hurrá! It works!

locki-cz, 2021-03-17:
great, fix works for me! I hope it will be soon fixed also for others via some updated version :)

OllisGit, 2021-04-11:
fyi: Currently I have some "private challenges to solve", thats the reason why It takes so long to release a new version, sorry!

Today I released a new development version (1.11.0). This version should fix this issue. Please test and give feedback!

Thx, in advance
Olli

melyux, 2021-05-04:
I’m on 1.12-dev1 and it still doesn’t work, with the same filament odometer error in the logs.

truglodite, 2021-05-05:
Same here, not working.

OllisGit, 2021-05-16:
Hi,

new version 1.3.0 is out and should fix some issues. Please test and give feedback.

 HINT: Still not working with current SpoolManager V1.3.4 you need to wait for V1.4.0, sorry!

Thx, in advance
Olli

OllisGit, 2021-05-24:
Try to use the lastest PJH-Plugin V1.14.0 and SpoolManger V1.4.0, that should work!

Please test and give feedback!
Olli

jkaessens, 2021-06-01:
Works perfectly! For me, at least. Thanks!

closed 2021-06-11 by OllisGit