OllisGit/OctoPrint-PrintJobHistory #119: Doesn’t seem to link to Spool Manager

Xiondewah, 2021-02-27:
Spool seems to be a text file rather than pull in from the Spool Manager plugin.

if this not an existing feature, it would be a very useful one 😁

bullshit, 2021-03-06:
I don't why, but the code for SpoolManager is commented out
https://github.com/OllisGit/OctoPrint-PrintJobHistory/blob/master/octoprint_PrintJobHistory/__init__.py#L136-L161

Xiondewah, 2021-03-07:
You're right.  I've no idea how to do anything with that so will have to leave it to the developer - hopefully!

bullshit, 2021-03-08:
I guess there are middle in the development. There are 2 compile errors in the master branch which is a little bit weird. idk
https://github.com/OllisGit/OctoPrint-PrintJobHistory/blob/master/octoprint_PrintJobHistory/__init__.py#L295
https://github.com/OllisGit/OctoPrint-PrintJobHistory/blob/master/octoprint_PrintJobHistory/__init__.py#L299

SM-26, 2021-03-08:
> I guess there are middle in the development. There are 2 compile errors in the master branch which is a little bit weird. idk
> https://github.com/OllisGit/OctoPrint-PrintJobHistory/blob/master/octoprint_PrintJobHistory/__init__.py#L295 (should be 'elif')
> https://github.com/OllisGit/OctoPrint-PrintJobHistory/blob/master/octoprint_PrintJobHistory/__init__.py#L299 (indentation wrong)

When I tryed out and installed your branch, after the print I get this:
```
2021-03-08 16:15:45,955 - octoprint.plugin - ERROR - Error while calling plugin PrintJobHistory
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 651, in on_event
    self._printJobFinished("success", payload)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 473, in _printJobFinished
    self._createAndAssignFilamentModel(self._currentPrintJobModel, payload)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 297, in _createAndAssignFilamentModel
    filemanentModel.usedLength = self._spoolManagerPluginImplementation.filamentOdometer.totalExtrusion[0]
AttributeError: 'SpoolmanagerPlugin' object has no attribute 'filamentOdometer'

```

Am I doing something wrong?

bullshit, 2021-03-09:
Okey i guess there are more bugs to find 🤦🏻

SM-26, 2021-03-09:
I don't know what I'm doing wrong but...
I've taking all of the PR and merged them into my Repo, then tried to install it
[image]

but for some reason. it thinks (again) that my python is out of date.
if anyone could tell me what I'm doing to cause this, it will be a really big helping hand.

OllisGit, 2021-04-11:
fyi: Currently I have some "private challenges to solve", thats the reason why It takes so long to release a new version, sorry!

Today I released a new development version (1.11.0). This version include the option for selecting the SpoolManager-Plugin for filament tracking.
**BUT you need the newest Version of SpoolManager 1.4.0+ and an other BUT this version is not released yet!**

So, please be patient it will take some time.

Thx, in advance 
Olli

bullshit, 2021-04-12:
Take your time! Good luck with your private challenges and in case you need help releasing the next version I and perhabs the community would gladly help you with that

Xiondewah, 2021-04-12:
Olli - sorry to be thick, but how does one install a version that hasn't been released (ie Spool Manager)?

OllisGit, 2021-05-24:
Try to use the lastest PJH-Plugin V1.14.0 and SpoolManger V1.4.0, that should work!

Please test and give feedback!
Olli

Xiondewah, 2021-05-25:
I'm afraid it didn't pick up the spool info from SpoolManager - and I'm on the latest version of everything.

OllisGit, 2021-05-25:
Hi @Xiondewah 

please update to SpoolManager V1.4.2, this should solve the issue.
Please test and give feedback.

Thx, in advance
Olli

SM-26, 2021-05-25:
It's working again for me with Spool Manager (1.4.2) & PrintJobHistory (1.14.0)
^.^

Xiondewah, 2021-05-25:
Hi Olli - sorry but I don't know how to do this as it doesn't come back as an available update within Plugin Manager. so I'm on 1.4.1

SM-26, 2021-05-25:
> Hi Olli - sorry but I don't know how to do this as it doesn't come back as an available update within Plugin Manager. so I'm on 1.4.1

the link for the latest is in the readme

just update from URL like this:
https://github.com/OllisGit/OctoPrint-SpoolManager/releases/download/1.4.2/master.zip

Xiondewah, 2021-05-25:
Thanks - updated - printing now :-)

closed 2021-05-25 by OllisGit

Xiondewah, 2021-05-25:
Yup - works like a charm thanks!