OllisGit/OctoPrint-PrintJobHistory #121: Filament Length being erased

ErikLindstrom42, 2021-03-08:
If I have Octoprint open on multiple machines it will report success on all of them, but only the originating one will record calculated filament length. Even if I accept and record the length on that machine, all the other machines will still have the print end dialog. The issue I am pointing out that even just closing those dialogs will overwrite the length to zero (or null) there is no option to keep the already recorded value.  It may be possible to close these tabs rather than closing the PJH popup and preserve the length, but I have not yet tested that.

OllisGit, 2021-04-11:
Hi @ErikLindstrom42,

juts for my understanding:
- you have one OP-Instance
- browsers on different machines also showing the OP-UI
- after print is finished the pjh-edit dialog appears on all machines and shows the filament length
- but if you press "close" on any of these dialog the length will be reseted to zero, right?

OllisGit, 2022-01-16:
closed, no feedback.

closed 2022-01-16 by OllisGit