OllisGit/OctoPrint-PrintJobHistory #123: M117 Snap Not Taking a Snapshot

alsmithson, 2021-03-09:
Possibly user error as the documentation is not very clear. My printer's end gcode includes lifting the nozzle 100mm at the end of the print to make it easier to get to the finished print. When PrintJobHistory takes a picture, the printed object is partly or wholly out of the frame. So I added the line _M117 Snapshot_ to the end gcode right before the _G0 Z100_ line. Then, in PrintJobHistory, I checked the box "Take camera snapshot if following gcode was sent" and put "M117 Snap" in the text box below it. However, PrintJobHistory still only takes a picture when the print job is completely finished and the gantry has raised up.

I have an Artillery Sidewinder X1 that uses a TFT controller. The printer itself ignores M117 messages, but shouldn't OctoPrint plugins see the code anyway?

OllisGit, 2021-04-11:
Hi @alsmithson,

if possible, please attach your gcode to this issue.

Did you try to add a "sleep" to your gcode (e.g. wait 1minute after moving to the end)? Maybe that helps.
```
M117 Snap
G4 P60000 
```

Btw. you can test the behaviour without wasting filament, if you use my [DryRun-Plugin](https://github.com/OllisGit/OctoPrint-DryRun)

MrSplint3r, 2021-04-14:
I have the same problem with 1.10 (haven't been able to test with 1.12 given incompatibility with currently released SpoolManager, sticking with 1.10 until then).  This is my End G-Code set up on Cura:

```
G91 ;Relative positioning
G1 E-2 F2700 ;Retract a bit
G1 E-2 Z0.2 F2400 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z10 ;Raise Z more
G90 ;Absolute positionning

M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed

; Make photo
M117 Snap
G4 P10000

G1 X0 Y{machine_depth} ;Present print
M84 X Y E ;Disable all steppers but Z
```

OllisGit, 2021-05-16:
Hi @alsmithson, @MrSplint3r,
I could not reproduce the issue. I used your script and a picture was taken.
Did you really activate the feature?
[image]
Or did you receive a error-popup like this?
[image]
Make sure that there is no space in front of the M117 message

Please take a log into the octoprint.log (or attach it here if your unsure).

OllisGit, 2021-05-16:
Hi,

new version 1.3.0 is out and maybe fix the issue.

Please test and give feedback.

Thx, in advance
Olli

Boetiebliksem, 2021-05-20:
Hi there.  I am also having the same issue where the M117 is seemingly ignored and the snapshot is only taken after the bed has moved to present the print.  I am new to using gcode so I am not 100% sure whether I entered it correctly in the end gcode in Cura.
This is a screenshot of the Gcode I downloaded from Octoprint after the print was done.  I did several tests and extended the 20 seconds to 60 seconds without success.  I also made sure the Snapshot is activated in the Plug in.
Please assist!
  
[image]
[image]

OllisGit, 2021-05-22:
Hi @Boetiebliksem,
in your gcode you use ```Tale snapshot``` with a lower s and in the text-expression field you enter it with an upper s. That is the reason why the plugin could not detect this command. Also make sure that behind you last letter are no spaces.

Boetiebliksem, 2021-05-23:
Hi @OllisGit,
Thank you for your reply, however, I fixed it as you explained above and made sure no spaces after the last letter in both gcode and text-expression and it still only takes a snapshot when the print bed is fully forward (presented).  What am I doing wrong?  
[image]
[image]

OllisGit, 2021-05-23:
hmmm...please take a look into octoprint.log
There must bie an entry like this:
```
M117 mesaage for taking snapshot detected. Try to capture image
```
When you can see this the image is taken by the expression. The log-entry also include a timestamp, maybe that helps as well.

I think the issue is the buffering of the gcode-lines. OP is listen on gcode where already sent to the printer. So did you already play around with the pause values. E.g. increase the pause in front of M117 and then only after M117.

Boetiebliksem, 2021-05-23:
Hi @OllisGit,
Yes, I found it in the log:
[image]

I have not tried to change the pause in front or after M117 again.  I will give it another go and keep you updated.

Boetiebliksem, 2021-05-26:
Hi @OllisGit 
No luck!  I upgraded to the latest version but it still snaps the image only after the bed has moved forward to present the print. I must be doing something wrong...  Any ideas where I can check?

Boetiebliksem, 2021-05-26:
Is there not perhaps another plugin that could interfere with this one?  Or perhaps a setting in Cura which is the slicer I am using.

OllisGit, 2021-05-27:
@Boetiebliksem playing around with the pause value and positioning the M117 code didn't help, right?
E.g. moving the M117 in front of the last layer.

Boetiebliksem, 2021-05-28:
I only played with the pause value but I did not move it around.  I will try it this afternoon, move the command just before the last layer start, and will give feedback.

alsmithson, 2021-06-08:
Even with 1.14.0, it still is not working. I tried your DryRun plugin to test multiple scenarios, always with the same result, It does not take the picture until the bed and gantry have moved to the final position. 

I get the proper message in the log (per your example above), I tried putting a G4 P10000 in front of the M117, behind it, and both (per your recommendation above). I have tried M117 Snapshot and M117 Snap in my gcode (per your example in the plugin itself). I disabled every 3rd party plugin except for PrintJobHistory and DryRun. I can't imagine what else I can try.

Here are my plugin settings:

[image]

Here is my gcode: 

; Begin SV IdeaMaker —printer— end gcode for Artillery Sidewinder X1
G92 E0 ; reset extruder position
G1 F1500 E-3 ; Retract 3 mm to prevent ooze.
; IdeaMaker raises the nozzle by 5mm at the end, so a z-hop isn’t needed here.
G4 P10000
M117 Snapshot
G4 P10000
G91 ; Set position axes to relative mode. 
G0 Z100 ; Raise the nozzle 100mm for easier cleaning and filament change.
M221 T0 S100 ; Set flow percentage in case something overrode it.
M104 S0 ; Turn off the nozzle heater.
M140 S0 ; Turn off the bed heater.
M106 S0 ; Turn off the fan.
G90 ; Set position axes to absolute mode. 
; G1 X0 Y200 F1000 ; Push the bed forward to prepare for part removal.
M84 ; Disable the motors.
M150 U255 ; Turn the LED green to indicate that the print is finished.
; End SV IdeaMaker —printer— end gcode for Artillery Sidewinder X1

alsmithson, 2021-07-12:
I upgraded to the latest OctoPi yesterday (I was already running the latest OctoPrint). The plugin will still not take a snapshot until the print is finished.

alsmithson, 2021-09-27:
What kind of test feedback are you waiting for?

OllisGit, 2021-11-13:
Because I have no idea, I raised a question in the OP-Forum: https://community.octoprint.org/t/take-image-via-gcode-command/38882

OllisGit, 2021-11-13:
Hi,
new release [1.15.0](https://github.com/OllisGit/OctoPrint-PrintJobHistory/releases/tag/1.15.0) is available and I changed the "snapshot-trigger" (OP-Hook), maybe that helps for you.

Please test and give me a feedback. 
Thx, in advance!
Olli

OllisGit, 2022-01-16:
fyi: In the next release the ```M118 approach`` from the forum-discussion is implemented. In a virtual-printer it works, but needs to be tested in a real-printer.

OllisGit, 2022-01-22:
Hi!

Well the M118 is implemented in the latest release 1.16.0, but I was not able to test the approach on a real printer.
During development with a virtual printer it works like expected.

So, please so kind kind, test and give me a feedback... a dry run print.

Thx,in advance
Olli

Boetiebliksem, 2022-01-22:
Hi @OllisGit!

Please supply sample code for taking the snapshot as I see we should use M118 now?  Do we still use "Take Snapshot" in the code?

Thanks!

Boetiebliksem, 2022-01-23:
Hi @OllisGit,

Please supply sample code as I am not sure about M118.  Would it look similar to the M117 command previously used? Apologies, but I am not as confide t with the codes as yet.

Thanks!

cmlaven, 2022-01-23:
@Boetiebliksem @OllisGit  

I used the sample code that is farther down in the plugin config. I tossed it in the end gcode in Cura. It has worked for my last 3 prints :)

[image]
[image]

OllisGit, 2022-01-24:
Hi @cmlaven thanks a lot for the feedback!

What printer and firmware do you use?

cmlaven, 2022-01-24:
Hi @OllisGit

No problem at all. I have an Ender 3 v2 running [Jyers ](https://github.com/Jyers/Marlin/releases/download/v2.0.1/E3V2-BLTouch-5x5-HS-v4.2.2-v2.0.1.bin)on a 4.2.2 board all hooked up via Octopi

locki-cz, 2022-02-19:
this is my end gcode ... it beeps wait beeps and no foto, prusa on last firmware with last octoprint, what i have wrong?

; Last edited 20200215
G4 ; wait
; Raise nozzle and present bed
M117 Printing complete
G4 ; wait
G92 E0 ; prepare to retract
G1 E-0.8 F3000; retract to avoid stringing
; Anti-stringing end wiggle
G91 ; use relative coordinates
G1 X-0.5 Y-0.5 F1200
G1 X1 Y1 F1200
G90 ; use absolute coordinates
{if layer_z < max_print_height}G1 Z{z_offset+min(layer_z+60, max_print_height)}{endif} ; Move print head up

//foto
M400 ;Finish moves
G4 P1000 ;Wait for no vibrations
M300 S440 P100 ;Playing a beep

M118 //action:pjhTakeSnapshot

G4 P5000 ;Waiting for Camera taken the Snapshot
M300 S440 P100 ;Playing a beep
//foto

G0 X0 Y210 F10200; present bed
; Reset print setting overrides
M200 D0 ; disable volumetric e
M220 S100 ; reset speed factor to 100%
M221 S100 ; reset extruder factor to 100%
M900 K0 ; reset linear acceleration
; Shut down printer
M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
M107 ; turn off fan
M84 ; disable motors
M300 S40 P10 ; chirp

OllisGit, 2022-03-18:
Hi @locki-cz,

please take a look into the ocotprint.log (and/or attach to this issue) to see some errors. Sometimes the M118 action-command is not recognized by the printer firmware.
Did you try the [DEPRECATED] M117 variant?

alsmithson, 2022-04-25:
The M118 method does work, though it triggers the camera before the print finishes. Adding G4 S5 before the M118 works so far. By the way, “M118 //action:pjhTakeSnapshot;” and “M118 A1 action:pjhTakeSnapshot;“ both work.

closed 2022-04-25 by alsmithson