OllisGit/OctoPrint-PrintJobHistory #125: "V1.11.0 will be released in the next days (CW05/2021)"

DyGyg, 2021-03-19:
Sorry if i'm pushing you...
but CW 05 is (or was):
05 | 01.02. | 02.02. | 03.02. | 04.02. | 05.02. | 06.02. | 07.02.

DaStivi, 2021-03-22:
yeah, i also was thinking to ask on ah status update ah couple of times... nothing too important as filamentmanager 1.7.0 works also fine for me...

Trashwarez, 2021-03-30:
Same here. Hope that we will get an update soon :-)

OllisGit, 2021-04-11:
fyi: Currently I have some "private challenges to solve", thats the reason why It takes so long to release a new version, sorry!

Today I release a new development version (1.11.0). If you find any issue please raise a new ticket and I will take a look on it.

Thx, in advance for testing
Olli

closed 2021-04-11 by OllisGit