OllisGit/OctoPrint-PrintJobHistory #127: Feature request: export print history

jmlg10, 2021-04-02:
Hi,

I'm trying to install PrintJobHstory as said in another "#issue"

I hope I shall find a way to make it running, but meanwhile I have one question.
It looks like it is possible to connect to the server and ask for exporting the file in CSV format.
But this is interactively.
I was wondering if there is any possibility to export it to a server location (or via eamil ) automatically after each print or each end of day?
Let me know.

Many Thanks

Jean-Michel

OllisGit, 2021-04-11:
Hi @jmlg10,
currently this feature is not "directly" planed!

The main purpose of the plugin is to grab the job data and store it to a sql-database. Storing the data to an other target (like cloud-storage, Google-Sheets, etc.)  is out of scope, BUT...

...one option I was thinking about is, to fire an event with all job informations to the Octoprint-Eventbus. 
With this feature it should be very easy to write your own plugin to push the job-data where ever you like.
Tracked in https://github.com/OllisGit/OctoPrint-PrintJobHistory/projects/1#card-32836851

closed 2021-04-11 by OllisGit

jmlg10, 2021-04-12:
> WARNING: This is an external email. Shall you have any doubt, report any suspicious e-mail by clicking on the "Report suspicious e-mail" button in Outlook.

Hi,

On the other hand, I want to generate a dashboard using the datas of the print history in PowerBI.
This software is able to analyze SQL databases, could it be possible to connect to your SQL server?

Let me know.

Cordialement / Best Regards

Jean-Michel LE GOFF | Responsable Recherches Numériques | Louis Vuitton Malletier | 2, Rue du Pont-Neuf, 75001 Paris, France | Phone: +33 155803048 | Mobile : +33 664405492 | Email address:  ***@***.***

OllisGit, 2021-05-16:
Hi @jmlg10,
currently it is not possible, because I am using a local filebased database.
It is planed to use a sql-server (e.g. postgres), but this is still in development.

jmlg10, 2021-05-17:
Ok,
I shall wait.
Let me know when it will be possible to try something.
Thanks