OllisGit/OctoPrint-PrintJobHistory #128: LOW: Update readme screenshots

OllisGit, 2021-04-11:


SM-26, 2021-04-12:
I'd be more than happy to help with this one.
just to take one care out of your head.

I'm guessing you are refereeing to 1.4.0?

OllisGit, 2021-05-16:
Hi @SM-26,
thanks for the offer. 

If you think there is any kind of documentation missing (readme or wiki) feel free to raise a pull request or just add the missing information in this ticket.

For V1.13.0 I only updated one screenshot.

SM-26, 2021-05-25:
I've made a commit with some new pictures.
https://github.com/SM-26/OctoPrint-PrintJobHistory/commit/5255800c8c9bd134ce855e83f2a46d05a4a96e1e

Made a new Wiki page
https://github.com/SM-26/OctoPrint-PrintJobHistory/wiki/Plugin-Setting-and-Setup/5b262b82c4b8243ca7c7e1056e496c98fe5bd000

Can I also ask you to make to wiki editable by anyone?
https://docs.github.com/en/communities/documenting-your-project-with-wikis/changing-access-permissions-for-wikis

closed 2022-02-02 by github-actions[bot]