OllisGit/OctoPrint-PrintJobHistory #130: Does Not install in OctoPi 0.18.0 octoprint 1.5.3

MarkFrTX, 2021-04-12:
New build of OctoPi, go to install my usual plugins but even though it stated successful install, my alert says:

Plugin management log
Install Unknown: 
A restart is needed for the changes to take effect.

After restart, it does not appear, and plugin manager lets me run install again.

Thanks for your help.

Installing plugin "PrintJobHistory" from https://github.com/OllisGit/OctoPrint-PrintJobHistory/releases/latest/download/master.zip...
/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmp16ngu59y/master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmp16ngu59y/master.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.7/site-packages (from Print-Job-History==1.11.0.dev0) (1.5.3)
Requirement already satisfied: pillow>=6.2.0<7.0.0 in /home/pi/oprint/lib/python3.7/site-packages (from Print-Job-History==1.11.0.dev0) (8.2.0)
Requirement already satisfied: peewee in /home/pi/oprint/lib/python3.7/site-packages (from Print-Job-History==1.11.0.dev0) (3.14.4)
Requirement already satisfied: sentry-sdk<1,>=0.15.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.19.1)
Requirement already satisfied: psutil<6,>=5.7 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (5.7.2)
Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.0.7)
Requirement already satisfied: semantic-version<3,>=2.8.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (2.8.5)
Requirement already satisfied: netaddr<1,>=0.7.19 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.8.0)
Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.1.0)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (2.0)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2020.09.23 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (2021.2.4)
Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (3.1.1)
Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.0.0)
Requirement already satisfied: cachelib<1,>=0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.1.1)
Requirement already satisfied: future<1,>=0.18.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.18.2)
Requirement already satisfied: zeroconf<0.25,>=0.24; python_version >= "3" in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.24.5)
Requirement already satisfied: sarge==0.1.5post0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.1.5.post0)
Requirement already satisfied: Jinja2<3,>=2.11.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (2.11.2)
Requirement already satisfied: werkzeug<2,>=1.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.0.1)
Requirement already satisfied: OctoPrint-FileCheck>=2020.08.07 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (2021.2.23)
Requirement already satisfied: watchdog==0.10.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.10.4)
Requirement already satisfied: websocket-client<1,>=0.57 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.57.0)
Requirement already satisfied: netifaces<1,>=0.10.9 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.10.9)
Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (2020.10.22)
Requirement already satisfied: pyserial<4,>=3.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (3.4)
Requirement already satisfied: blinker<2,>=1.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.4)
Requirement already satisfied: flask<2,>=1.1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.1.2)
Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.5.0)
Requirement already satisfied: feedparser<7,>=6.0.2; python_version >= "3" in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (6.0.2)
Requirement already satisfied: pylru<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.2.0)
Requirement already satisfied: tornado==5.1.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (5.1.1)
Requirement already satisfied: wrapt<2,>=1.12.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.12.1)
Requirement already satisfied: PyYAML<6,>=5.3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (5.3.1)
Requirement already satisfied: frozendict<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.2)
Requirement already satisfied: Click<8,>=7.1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (7.1.2)
Requirement already satisfied: requests<3,>=2.23.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (2.24.0)
Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.6.0)
Requirement already satisfied: emoji<1,>=0.5.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.6.0)
Requirement already satisfied: unidecode<0.05,>=0.04.14 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.4.21)
Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.1.1)
Requirement already satisfied: rsa==4.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (4.0)
Requirement already satisfied: urllib3>=1.10.0 in /home/pi/oprint/lib/python3.7/site-packages (from sentry-sdk<1,>=0.15.1->OctoPrint->Print-Job-History==1.11.0.dev0) (1.25.11)
Requirement already satisfied: certifi in /home/pi/oprint/lib/python3.7/site-packages (from sentry-sdk<1,>=0.15.1->OctoPrint->Print-Job-History==1.11.0.dev0) (2020.6.20)
Requirement already satisfied: webassets>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from Flask-Assets<3,>=2.0->OctoPrint->Print-Job-History==1.11.0.dev0) (2.0)
Requirement already satisfied: setuptools>=36 in /home/pi/oprint/lib/python3.7/site-packages (from markdown<3.2,>=3.1->OctoPrint->Print-Job-History==1.11.0.dev0) (50.3.2)
Requirement already satisfied: Babel>=2.3 in /home/pi/oprint/lib/python3.7/site-packages (from Flask-Babel<2,>=1.0->OctoPrint->Print-Job-History==1.11.0.dev0) (2.8.0)
Requirement already satisfied: pytz in /home/pi/oprint/lib/python3.7/site-packages (from Flask-Babel<2,>=1.0->OctoPrint->Print-Job-History==1.11.0.dev0) (2020.1)
Requirement already satisfied: ifaddr in /home/pi/oprint/lib/python3.7/site-packages (from zeroconf<0.25,>=0.24; python_version >= "3"->OctoPrint->Print-Job-History==1.11.0.dev0) (0.1.7)
Requirement already satisfied: pathtools>=0.1.1 in /home/pi/oprint/lib/python3.7/site-packages (from watchdog==0.10.4->OctoPrint->Print-Job-History==1.11.0.dev0) (0.1.2)
[... cut ...]
Created wheel for Print-Job-History: filename=Print_Job_History-1.11.0.dev0-py3-none-any.whl size=274188 sha256=edc0e375fc55bc862367c25e2409ff7340e730abc654828e357543af02fc633c
Stored in directory: /tmp/pip-ephem-wheel-cache-v_78_9bi/wheels/e2/4b/64/f9428901cbf4f8af81f2af9017e1aa48815e99225336df6ede
Successfully built Print-Job-History
Installing collected packages: Print-Job-History
Attempting uninstall: Print-Job-History
Found existing installation: Print-Job-History 1.11.0.dev0
Uninstalling Print-Job-History-1.11.0.dev0:
Successfully uninstalled Print-Job-History-1.11.0.dev0
Successfully installed Print-Job-History-1.11.0.dev0
Done!

condraka, 2021-04-12:
Also have same issue.  Disappeared on install of Octopi 0.18.0  Python 3.7.3  Octoprint 1.5.3  cannot reinstall.  (it already exsists but will not display)..  Same as above poster.

Doprintityourself, 2021-04-12:
Same here just wanted to migrate to 0.18.
After installing the backup form my former 0.17 pi4 it wont appear
cleaning yaml and plugin by hand 
every time I attempt to install it it runs through but it is not installed after that:

Installiere Plugin "PrintJobHistory" von https://github.com/OllisGit/OctoPrint-PrintJobHistory/releases/latest/download/master.zip...
/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpjc_xi11s/master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmpjc_xi11s/master.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.7/site-packages (from Print-Job-History==1.11.0.dev0) (1.5.3)
Requirement already satisfied: pillow>=6.2.0<7.0.0 in /home/pi/oprint/lib/python3.7/site-packages (from Print-Job-History==1.11.0.dev0) (8.2.0)
Requirement already satisfied: peewee in /home/pi/oprint/lib/python3.7/site-packages (from Print-Job-History==1.11.0.dev0) (3.14.4)
Requirement already satisfied: netifaces<1,>=0.10.9 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.10.9)
Requirement already satisfied: wrapt<2,>=1.12.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.12.1)
Requirement already satisfied: future<1,>=0.18.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.18.2)
Requirement already satisfied: pylru<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.2.0)
Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.5.0)
Requirement already satisfied: watchdog==0.10.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.10.4)
Requirement already satisfied: frozendict<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.2)
Requirement already satisfied: unidecode<0.05,>=0.04.14 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.4.21)
Requirement already satisfied: flask<2,>=1.1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.1.2)
Requirement already satisfied: cachelib<1,>=0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.1.1)
Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.0.7)
Requirement already satisfied: zeroconf<0.25,>=0.24 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.24.5)
Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (3.1.1)
Requirement already satisfied: Jinja2<3,>=2.11.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (2.11.2)
Requirement already satisfied: semantic-version<3,>=2.8.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (2.8.5)
Requirement already satisfied: Click<8,>=7.1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (7.1.2)
Requirement already satisfied: emoji<1,>=0.5.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.6.0)
Requirement already satisfied: OctoPrint-FileCheck>=2020.08.07 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (2021.2.23)
Requirement already satisfied: sarge==0.1.5post0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.1.5.post0)
Requirement already satisfied: feedparser<7,>=6.0.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (6.0.2)
Requirement already satisfied: websocket-client<1,>=0.57 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.57.0)
Requirement already satisfied: pyserial<4,>=3.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (3.5)
Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.1.1)
Requirement already satisfied: requests<3,>=2.23.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (2.25.1)
Requirement already satisfied: tornado==5.1.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (5.1.1)
Requirement already satisfied: blinker<2,>=1.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.4)
Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (2020.11.13)
Requirement already satisfied: netaddr<1,>=0.7.19 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.8.0)
[... cut ...]
Created wheel for Print-Job-History: filename=Print_Job_History-1.11.0.dev0-py3-none-any.whl size=274190 sha256=9affd5974b1672586cb8a67431d77e071fdd5019b19b91893b70824f2ee8da2c
Stored in directory: /tmp/pip-ephem-wheel-cache-kdvzm91k/wheels/c0/45/f2/fa19f2ea8bb7bbcd5a589d41bd6c9a42bd9eff6a0ce82631c5
Successfully built Print-Job-History
Installing collected packages: Print-Job-History
Attempting uninstall: Print-Job-History
Found existing installation: Print-Job-History 1.11.0.dev0
Uninstalling Print-Job-History-1.11.0.dev0:
Successfully uninstalled Print-Job-History-1.11.0.dev0
Successfully installed Print-Job-History-1.11.0.dev0
Fertig!

EldritchE, 2021-04-12:
Also have same issue

OllisGit, 2021-04-12:
Duplicated-Issue, see #131 
Hi,

sorry, didn't saw that one new file was not pushed to git.

In the new version 1.11.1-dev the SlicerSettingsService.py is now included.

Please test and give feedback.

Thx, in advance
Olli

Doprintityourself, 2021-04-12:
Great thanks, 
Works for me here.

I could install and see the plugin now, but it keeps on throwing up the update dialog even if already installed
[image]
After First Test everything went fine.
The finished print appeared, but I realized that Filamentmanger is not working with Printjobhistory!

Is there any news about when Spoolmanger will come with the external database?
Would really appreciate that!

condraka, 2021-04-12:
See yesterday’s email about this.  It’s in process 

Carl E. Ondraka

MarkFrTX, 2021-04-12:
Hi Olli,

I was able to get it to install on my test pi. THough It is giving me an 
alert...

So it is working but thinks it needs to be updated.  I tried Update now, 
and it just popped up alert again after completing.

Thanks for you help!
Mark

OctoPi 0.18.0 octoprint 1.5.3 (#130)

OllisGit, 2021-04-13:
Hi,

in the latest release I change the version numbering to the correct pattern. Now the new version 1.12.dev1 should work without "endless-popup"

Please test and give feedback.

Thx, in advance
Olli

MarkFrTX, 2021-04-13:
Just did a ground up install and PrintJobHistory plugin installed 
without a hitch.

Thanks for you diligence in getting this resolved!!!
M

closed 2021-04-14 by MarkFrTX