OllisGit/OctoPrint-PrintJobHistory #131: Error loading plugin PrintJobHistory on FreeBSD 12.2 jail - module not found SlicerSettingsService

NguyenTanBao, 2021-04-12:
Hi,

I have Octoprint running inside FreeBSD 12.2 jail. Installing PrintJobHistory from PluginManager says Install Unknown but it's successfully in the plugin_pluginmanager_console.log

`2021-04-11 22:41:53,780   /usr/home/octoprint/octoprint/venv/bin/python3.7 -m pip --disable-pip-version-check install file:///tmp/tmp9xhaio8q/master.zip --no-cache-dir
2021-04-11 22:41:54,896 > Processing /tmp/tmp9xhaio8q/master.zip
2021-04-11 22:41:56,383 > Requirement already satisfied: OctoPrint in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from Print-Job-History==1.11.0.dev0) (1.5.3)
2021-04-11 22:41:56,384 > Requirement already satisfied: pillow>=6.2.0<7.0.0 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from Print-Job-History==1.11.0.dev0) (8.2.0)
2021-04-11 22:41:56,384 > Requirement already satisfied: peewee in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from Print-Job-History==1.11.0.dev0) (3.14.4)
2021-04-11 22:41:56,384 > Requirement already satisfied: markupsafe<2.0,>=1.1 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.1.1)
2021-04-11 22:41:56,384 > Requirement already satisfied: unidecode<0.05,>=0.04.14 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.4.21)
2021-04-11 22:41:56,384 > Requirement already satisfied: zeroconf<0.25,>=0.24 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.24.5)
2021-04-11 22:41:56,384 > Requirement already satisfied: Click<8,>=7.1.2 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (7.1.2)
2021-04-11 22:41:56,385 > Requirement already satisfied: OctoPrint-FirmwareCheck>=2020.09.23 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (2021.2.4)
2021-04-11 22:41:56,385 > Requirement already satisfied: psutil<6,>=5.7 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (5.8.0)
2021-04-11 22:41:56,385 > Requirement already satisfied: werkzeug<2,>=1.0.1 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.0.1)
2021-04-11 22:41:56,385 > Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.7.0)
2021-04-11 22:41:56,385 > Requirement already satisfied: future<1,>=0.18.2 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.18.2)
2021-04-11 22:41:56,385 > Requirement already satisfied: Jinja2<3,>=2.11.2 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (2.11.3)
2021-04-11 22:41:56,385 > Requirement already satisfied: watchdog==0.10.4 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.10.4)
2021-04-11 22:41:56,385 > Requirement already satisfied: blinker<2,>=1.4 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.4)
2021-04-11 22:41:56,385 > Requirement already satisfied: tornado==5.1.1 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (5.1.1)
2021-04-11 22:41:56,386 > Requirement already satisfied: Flask-Babel<2,>=1.0 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.0.0)
2021-04-11 22:41:56,386 > Requirement already satisfied: feedparser<7,>=6.0.2 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (6.0.2)
2021-04-11 22:41:56,386 > Requirement already satisfied: pylru<2,>=1.2 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.2.0)
2021-04-11 22:41:56,386 > Requirement already satisfied: netaddr<1,>=0.7.19 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.8.0)
2021-04-11 22:41:56,386 > Requirement already satisfied: semantic-version<3,>=2.8.5 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (2.8.5)
2021-04-11 22:41:56,386 > Requirement already satisfied: regex!=2018.11.6 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (2020.11.13)
2021-04-11 22:41:56,386 > Requirement already satisfied: sarge==0.1.5post0 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.1.5.post0)
2021-04-11 22:41:56,386 > Requirement already satisfied: filetype<2,>=1.0.7 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.0.7)
2021-04-11 22:41:56,386 > Requirement already satisfied: cachelib<1,>=0.1 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.1.1)
2021-04-11 22:41:56,386 > Requirement already satisfied: netifaces<1,>=0.10.9 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.10.9)
2021-04-11 22:41:56,387 > Requirement already satisfied: flask<2,>=1.1.2 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.1.2)
2021-04-11 22:41:56,387 > Requirement already satisfied: Flask-Login<0.6,>=0.5 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.5.0)
2021-04-11 22:41:56,387 > Requirement already satisfied: OctoPrint-FileCheck>=2020.08.07 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (2021.2.23)
2021-04-11 22:41:56,387 > Requirement already satisfied: markdown<3.2,>=3.1 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (3.1.1)
2021-04-11 22:41:56,387 > Requirement already satisfied: Flask-Assets<3,>=2.0 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (2.0)
2021-04-11 22:41:56,387 > Requirement already satisfied: requests<3,>=2.23.0 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (2.25.1)
2021-04-11 22:41:56,387 > Requirement already satisfied: pyserial<4,>=3.4 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (3.5)
2021-04-11 22:41:56,387 > Requirement already satisfied: sentry-sdk<1,>=0.15.1 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.20.3)
2021-04-11 22:41:56,387 > Requirement already satisfied: PyYAML<6,>=5.3.1 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (5.4.1)
2021-04-11 22:41:56,388 > Requirement already satisfied: wrapt<2,>=1.12.1 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.12.1)
2021-04-11 22:41:56,388 > Requirement already satisfied: rsa==4.0 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (4.0)
2021-04-11 22:41:56,388 > Requirement already satisfied: emoji<1,>=0.5.4 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.6.0)
2021-04-11 22:41:56,388 > Requirement already satisfied: websocket-client<1,>=0.57 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (0.58.0)
2021-04-11 22:41:56,388 > Requirement already satisfied: itsdangerous<2,>=1.1.0 in /usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages (from OctoPrint->Print-Job-History==1.11.0.dev0) (1.1.0)
[... cut ...]
    from octoprint_PrintJobHistory.api import TransformPrintJob2JSON, TransformSlicerSettings2JSON
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/octoprint_PrintJobHistory/api/TransformSlicerSettings2JSON.py", line 4, in <module>
    from octoprint_PrintJobHistory.services.SlicerSettingsService import SlicerSettingsService
ModuleNotFoundError: No module named 'octoprint_PrintJobHistory.services.SlicerSettingsService'
`

Is anyone hitting the same issue? Any recommendations? Thanks

alexscaliante, 2021-04-12:
I'm seeing the same problem after updating yesterday from 1.10 to V1.11.0-dev. Seems to have been introduced in this version. This also resulted in the plugin "disappearing" (not being able to initialize) as described in https://github.com/OllisGit/OctoPrint-PrintJobHistory/issues/130

Quintin9801, 2021-04-12:
I am also having this exact problem with Octoprint 1.5.3 running on Ubuntu 20.04.2 LTS. Reinstalled 1.10 manually to fix.

DyGyg, 2021-04-12:
@OllisGit  i think you forget to commit octoprint_PrintJobHistory/services/SlicerSettingsService

NguyenTanBao, 2021-04-12:
> I am also having this exact problem with Octoprint 1.5.3 running on Ubuntu 20.04.2 LTS. Reinstalled 1.10 manually to fix.

I didn't know you could uninstall/install manually. I searched around and haven't found a good way to do it, can you share any documentation link out there? Thanks

DyGyg, 2021-04-12:
@NguyenTanBao just install from URL
https://github.com/OllisGit/OctoPrint-PrintJobHistory/releases/download/1.10.0/master.zip

OllisGit, 2021-04-12:
Hi,

sorry, didn't saw that one new file was not pushed to git.

In the new version 1.11.1-dev the SlicerSettingsService.py is now included.

Please test and give feedback.

Thx, in advance
Olli

Quintin9801, 2021-04-12:
> Hi,
> 
> sorry, didn't saw that one new file was not pushed to git.
> 
> In the new version 1.11.1-dev the SlicerSettingsService.py is now included.
> 
> Please test and give feedback.
> 
> Thx, in advance
> Olli
@OllisGit Version 1.11.1-dev works as expected, after updating the plugin did not disappear like in 1.10

NguyenTanBao, 2021-04-13:
I no longer see original issue. I hit some other issues and have created a different issue for that. Thanks

closed 2021-04-13 by OllisGit