OllisGit/OctoPrint-PrintJobHistory #133: Use PEP440 naming so dev versions are automatically ignored

WoLpH, 2021-04-12:
Just a little tip, if you change the version from `1.11.1-dev` to `1.11.dev1` it should automatically be recognized as a development release.
https://www.python.org/dev/peps/pep-0440/#developmental-releases

That way you won't bother the general user with an update notification :)

Thank you for your great work!

OllisGit, 2021-04-13:
Hi @WoLpH,

thanks a lot for the hint about the correct naming pattern.
Now it should fit.

Please test and give a feedback.

Thx, in advance
Olli

WoLpH, 2021-04-13:
Hi Olli,

It seems to work. But it's still offering the update to the previous dev release. So you'll have to pull that release :)

WoLpH, 2021-04-16:
It seems that Octoprint doesn't understand semantic versioning for plugins (even though they use it themselves).

Apparently to get Octoprint to understand it you need to override the `update_hook`: https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/da97332eb1fecc41f3a0b21a92a8f088f3be9555/octoprint_firmwareupdater/__init__.py#L709-L746

OllisGit, 2021-05-16:
Hi @WoLpH,

thanks for the hint.

The new release 1.13.0 use the "normal" versioning and should solve the issue.
Please test and give feedback.

Thx, in advance
Olli

WoLpH, 2021-05-17:
It seems to work great for me @OllisGit  :)

closed 2021-05-17 by WoLpH