OllisGit/OctoPrint-PrintJobHistory #135: 1.11.1-dev doesn't record the print

NguyenTanBao, 2021-04-13:
Hi,

I'm using Spool Manager (not Filament  Management)
With 1.10, at the end of the print, it pops up a dialog and ask to add a print record (to update used length, used weight.. because I think 1.10 didn't work with Spool Manager to pull data.

With 1.11.1-dev, there's no popup dialog to fill in the data (I would expected it automatically gets from Spool Manager) and also there's no new record in the history. One thing I did differently is after issue the print from OctoPrint, I closed the browser. I would expact PrintJobHistory should still record the print even user closes the browser.

Here is my configuration:
[image]

OllisGit, 2021-04-13:
Hi @NguyenTanBao,

well the current situation is not the best. I released a new PJH with a SpoolManager dependency which doesn't exists yet ( I am still testing and fixing bug).

The reason for your issue is that an error occurs and the print job is not captured. Take a look into the octoprint.log (download able via settings-menu)

There is no version-check against the SpoolManger-Plugin. You use an incompatible version of spool-manager (of course, because the compatible version is not available ;-)

Please deactivate the SpoolManager plugin or install the FilamentManager and switch to this tracking-plugin.
After the the Printjobs should be captured again.

dieseltravis, 2021-04-13:
I have this same issue, so I switched back to FilamentManager, and it still didn't show the edit dialog at the end of the print but it did seem to save it correctly.

NguyenTanBao, 2021-04-13:
Got it, thanks. I thought FilamentManager is obsolete and everyone moves to SpoolManager. That's why I chose SpoolManager. I'll hang around a bit as migrating data from SpoolManager back to FilamentManager is not fun.

NguyenTanBao, 2021-04-13:
And here is the log for your confirmation:

`2021-04-13 10:47:43,477 - octoprint - ERROR - Exception on /plugin/PrintJobHistory/storePrintJob/1 [PUT]
Traceback (most recent call last):
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/octoprint_PrintJobHistory/api/PrintJobHistoryAPI.py", line 304, in put_printjob
    self._updatePrintJobFromJson(printJobModel, jsonData)
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/octoprint_PrintJobHistory/api/PrintJobHistoryAPI.py", line 63, in _updatePrintJobFromJson
    filamentModel.profileVendor = self._getValueFromJSONOrNone("spoolVendor", jsonData)
AttributeError: 'list' object has no attribute 'profileVendor'
2021-04-13 10:47:43,480 - tornado.access - ERROR - 500 PUT /plugin/PrintJobHistory/storePrintJob/1 (192.168.1.25) 14.30ms
2021-04-13 10:47:45,878 - octoprint - ERROR - Exception on /plugin/PrintJobHistory/storePrintJob/1 [PUT]
Traceback (most recent call last):
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/octoprint_PrintJobHistory/api/PrintJobHistoryAPI.py", line 304, in put_printjob
    self._updatePrintJobFromJson(printJobModel, jsonData)
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/octoprint_PrintJobHistory/api/PrintJobHistoryAPI.py", line 63, in _updatePrintJobFromJson
    filamentModel.profileVendor = self._getValueFromJSONOrNone("spoolVendor", jsonData)
AttributeError: 'list' object has no attribute 'profileVendor'
2021-04-13 10:47:45,880 - tornado.access - ERROR - 500 PUT /plugin/PrintJobHistory/storePrintJob/1 (192.168.1.25) 10.69ms
2021-04-13 10:47:49,609 - octoprint - ERROR - Exception on /plugin/PrintJobHistory/storePrintJob/1 [PUT]
Traceback (most recent call last):
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/octoprint_PrintJobHistory/api/PrintJobHistoryAPI.py", line 304, in put_printjob
    self._updatePrintJobFromJson(printJobModel, jsonData)
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/octoprint_PrintJobHistory/api/PrintJobHistoryAPI.py", line 63, in _updatePrintJobFromJson
    filamentModel.profileVendor = self._getValueFromJSONOrNone("spoolVendor", jsonData)
[... cut ...]
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/octoprint_PrintJobHistory/api/PrintJobHistoryAPI.py", line 304, in put_printjob
    self._updatePrintJobFromJson(printJobModel, jsonData)
  File "/usr/home/octoprint/octoprint/venv/lib/python3.7/site-packages/octoprint_PrintJobHistory/api/PrintJobHistoryAPI.py", line 63, in _updatePrintJobFromJson
    filamentModel.profileVendor = self._getValueFromJSONOrNone("spoolVendor", jsonData)
AttributeError: 'list' object has no attribute 'profileVendor'
2021-04-13 10:47:50,534 - tornado.access - ERROR - 500 PUT /plugin/PrintJobHistory/storePrintJob/1 (192.168.1.25) 10.09ms`

papinist, 2021-04-18:
I have the same behaviour with 1.12.dev1 - no save screen and no entry in history (I'm using spoolmanager)

Quintin9801, 2021-04-19:
> I have the same behaviour with 1.12.dev1 - no save screen and no entry in history (I'm using spoolmanager)

I am experiencing this same issue running OctoPrint 1.5.3, Python 3.8.5, Ubuntu 20.04.2 LTS

melyux, 2021-05-04:
Still happening. Migrated to Spool Manager from Filament Manager before trying PrintJobHistory, so never have been able to see this plugin work.

OllisGit, 2021-05-16:
Hi,

new version 1.3.0 is out and should fix the issue.

Please test and give feedback.

Thx, in advance
Olli

OllisGit, 2021-05-24:
Try to use the lastest PJH-Plugin V1.14.0 and SpoolManger V1.4.0, that should work!

Please test and give feedback!
Olli

PoloClive10, 2021-05-27:
Hi,

I am running PJH V1.14.0 and SpoolManager V1.4.1 and PJH hasn't captured any of the prints that I have run.

Please advise if there is something that I can do to fix this.

Thanks
Clive

OllisGit, 2021-05-27:
@PoloClive10, please try the latest release of SpoolManager V1.4.3: https://github.com/OllisGit/OctoPrint-SpoolManager/releases

closed 2021-10-30 by OllisGit