OllisGit/OctoPrint-PrintJobHistory #138: Can't Save Edit

jimml1, 2021-04-15:
Everything was working fine with v1.10. When I upgraded to v1.12, I can no longer save an edit. When I click the Save button nothing happens.

Installed Plug-ins
Cura Thumbnails
Dashboard
DisplayLayerProgress
Exclude Region
Filament Manager
Firmware Check
Floating Navbar
Preheat Button
Print Job History
PrintTimeGenius Plugin
Printer Dialogs
Printer Notifications
PrusaSlicer Thumbnails
Spool Manager
Themeify
Virtual Printer

wiglot, 2021-04-24:
Good Day. 

Same problem here. This is my first install of Print Job history plugin and never saved any history. Looking at logs, but didn't find any trace of plugin.

Plugins installed:
[{"name": "MQTT Subscribe", "key": "mqttsubscribe"}, 
{"name": "Telegram Notifications", "key": "telegram"}, 
{"name": "PrintJobHistory", "key": "PrintJobHistory"}, 
{"name": "Bed Visualizer", "key": "bedlevelvisualizer"}, 
{"name": "Cancel Objects", "key": "cancelobject"}, 
{"name": "Filament Manager", "key": "filamentmanager"}, 
{"name": "SpoolManager Plugin", "key": "SpoolManager"}, 
{"name": "Exclude Region", "key": "excluderegion"}, 
{"name": "MQTT", "key": "mqtt"}, 
{"name": "PrusaSlicer Thumbnails", "key": "prusaslicerthumbnails"}, 
{"name": "HomeAssistant Discovery", "key": "homeassistant"}, 
{"name": "Navbar Temperature Plugin", "key": "navbartemp"}]

jp-phys-NEIU, 2021-04-25:
Hello. I have had the same problem since Feb 2 '21. Octoprint 1.5.3,  PrintJobHistory 1.12. Print Job History records a job in the history with the right date and filename. However, I can't add the filament type, or add an image later (my setup doesn't have a camera, I manually upload a shot from my phone). The editing process goes through the motions but "Save" doesn't save the information. Saved images/data from 2/2/21 and before is still there.

OllisGit, 2021-05-16:
Hi,

new version 1.3.0 is out and should fix the issue.

Please test and give feedback.

Thx, in advance
Olli

jp-phys-NEIU, 2021-05-16:
Thanks for the update!
Installed 1.3 along with OctoPrint 1.6.1. 
Clicking the PrintJobHistory Save button now works - I can now enter text in spool, spool vendor, and other fields and the changes are successfully saved. Before the upgrade to 1.3, it appeared that I could save images, but clicking the Save button (instead of just Close) after Upload wiped the images out. Now images remain in the database after clicking Save.

OllisGit, 2021-05-17:
Hi @jp-phys-NEIU,
thanks for the fast response!!!

BR
Olli 
BTW. Images were stored in filesystem and not in the database ;-)

closed 2021-05-17 by OllisGit

jimml1, 2021-05-17:
Everything is working great with v1.3
Thanks for the update.

Thanks
Jim