OllisGit/OctoPrint-PrintJobHistory #139: The plug-in disappeared

santanig5, 2021-04-16:
Greetings,
I've been using the plug-in for a short time now, after my last print job I was not prompted for the job info and when I went to click on the history tab it was gone!
I reinstalled the plug-in and restored the db but nothing shows in the history list, yet if I click on stats it shows 72 print jobs.
Any help would be appriciated.

OllisGit, 2021-05-16:
Hi @santanig5 ,

new version 1.3.0 is out and should fix the issue.

Please test and give feedback.

Thx, in advance
Olli

santanig5, 2021-05-17:
unfortunately no, stats show 25 print jobs, however the tab is blank.

[image]
[image]

OllisGit, 2021-05-17:
Hi @santanig5,

please provide me your database so that I can reproduce the issue on my system.
[image]

santanig5, 2021-05-18:
per your request

[printJobHistory.zip](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/6497602/printJobHistory.zip)

OllisGit, 2021-05-18:
Hii @santanig5 ,
thanks for the database, very helpfull.
I could fix the issue in the latest version 1.13.1

Please test and give feedback.

Thx, in advance
Olli

santanig5, 2021-05-18:
I have installed 1.13.1 and will report back, in the mean time can you assist me in appending the db with the log I was attempting to maintain while this issue existed? When I do I get a screen full of errors.
See attached, two formats same file.

Thank you

santanig5, 2021-05-19:
Seems to be working correctly now with 1.13.1. Thank You!
Is there anyway to add those jobs I manually added to the csv file and import?

OllisGit, 2021-05-19:
Hi @santanig5,

there is an importer in the plugin settings:
[image]

closed 2021-05-19 by OllisGit

santanig5, 2021-05-19:
yes, but it gives lots of errors and fails to import

[image]<https://user-images.githubusercontent.com/24205767/118867055-f3f56f00-b8e2-11eb-9ad2-957e80fc06ac.png>

OllisGit, 2021-05-22:
@santanig5 issue is tracked here: #146 
Btw. just saying there are a lot of errors is not very helpfull. Next time, please provide more informations like:  a screenshot, exact error message, octoprint.log, database, ....

santanig5, 2021-05-23:
Sure, sorry. you have the DB already and I sent you the .csv file I thought that would be enough.
The initial errors are as follows, I'm sure it's because I tried to manually update the CSV file without all of the info required.

Some error(s) occurs during parsing! No jobs imported!
[2] Mandatory value for column 'Start Datetime [dd.mm.yyyy hh:mm]' is missing!
[2]Error parsing value 'White #5' for field 'Spool Name': dict values object is not subscriptable
[2]Error parsing value 'PLA+' for field 'Material': dict values object is not subscriptable
[2]Error parsing value '2000' for field 'Used Length [mm]': dict values object is not subscriptable
[2]Error parsing value '83540' for field 'Calculated Length [mm]': dict values object is not subscriptable
[2]Error parsing value '247' for field 'Used Weight [g]': dict values object is not subscriptable
[2]Error parsing value '12' for field 'Used Filament Cost': dict values object is not subscriptable
[3] Mandatory value for column 'Start Datetime [dd.mm.yyyy hh:mm]' is missing!
[3]Error parsing value 'Zyltech Clear' for field 'Spool Name': dict values object is not subscriptable
[3]Error parsing value 'PLA' for field 'Material': dict values object is not subscriptable
[3]Error parsing value '3510' for field 'Used Length [mm]': dict values object is not subscriptable
[3]Error parsing value '3490' for field 'Calculated Length [mm]': dict values object is not subscriptable
[3]Error parsing value '9' for field 'Used Weight [g]': dict values object is not subscriptable
[3]Error parsing value '19.99' for field 'Used Filament Cost': dict values object is not subscriptable
[4] Mandatory value for column 'Start Datetime [dd.mm.yyyy hh:mm]' is missing!
[4]Error parsing value 'White #5' for field 'Spool Name': dict values object is not subscriptable
[4]Error parsing value 'PLA+' for field 'Material': dict values object is not subscriptable
[4]Error parsing value '1140' for field 'Used Length [mm]': dict values object is not subscriptable
[4]Error parsing value '1126' for field 'Calculated Length [mm]': dict values object is not subscriptable
[4]Error parsing value '4' for field 'Used Weight [g]': dict values object is not subscriptable
[4]Error parsing value '12' for field 'Used Filament Cost': dict values object is not subscriptable
[5] Mandatory value for column 'Start Datetime [dd.mm.yyyy hh:mm]' is missing!
[5]Error parsing value 'White #5' for field 'Spool Name': dict values object is not subscriptable
[5]Error parsing value 'PLA+' for field 'Material': dict values object is not subscriptable
[5]Error parsing value '8500' for field 'Used Length [mm]': dict values object is not subscriptable
[5]Error parsing value '8500' for field 'Calculated Length [mm]': dict values object is not subscriptable
[5]Error parsing value '3' for field 'Used Weight [g]': dict values object is not subscriptable
[5]Error parsing value '12' for field 'Used Filament Cost': dict values object is not subscriptable
[6] Mandatory value for column 'Start Datetime [dd.mm.yyyy hh:mm]' is missing!
[6]Error parsing value 'Black' for field 'Spool Name': dict values object is not subscriptable
[6]Error parsing value 'TPU' for field 'Material': dict values object is not subscriptable
[6]Error parsing value '7490000' for field 'Used Length [mm]': dict values object is not subscriptable
[6]Error parsing value '7490000' for field 'Calculated Length [mm]': dict values object is not subscriptable
[6]Error parsing value '23' for field 'Used Weight [g]': dict values object is not subscriptable
[6]Error parsing value '19.99' for field 'Used Filament Cost': dict values object is not subscriptable
[7] Mandatory value for column 'Start Datetime [dd.mm.yyyy hh:mm]' is missing!
[7]Error parsing value 'White #5' for field 'Spool Name': dict values object is not subscriptable
[7]Error parsing value 'PLA+' for field 'Material': dict values object is not subscriptable
[7]Error parsing value '775000' for field 'Used Length [mm]': dict values object is not subscriptable
[7]Error parsing value '775000' for field 'Calculated Length [mm]': dict values object is not subscriptable
[7]Error parsing value '23' for field 'Used Weight [g]': dict values object is not subscriptable
[7]Error parsing value '12' for field 'Used Filament Cost': dict values object is not subscriptable
[8] Mandatory value for column 'Start Datetime [dd.mm.yyyy hh:mm]' is missing!
[8]Error parsing value 'White #5' for field 'Spool Name': dict values object is not subscriptable
[8]Error parsing value 'PLA+' for field 'Material': dict values object is not subscriptable
[8]Error parsing value '160600' for field 'Used Length [mm]': dict values object is not subscriptable
[8]Error parsing value '160600' for field 'Calculated Length [mm]': dict values object is not subscriptable
[8]Error parsing value '48' for field 'Used Weight [g]': dict values object is not subscriptable
[8]Error parsing value '12' for field 'Used Filament Cost': dict values object is not subscriptable
[9] Mandatory value for column 'Start Datetime [dd.mm.yyyy hh:mm]' is missing!
[9]Error parsing value 'White #5' for field 'Spool Name': dict values object is not subscriptable
[9]Error parsing value 'PLA+' for field 'Material': dict values object is not subscriptable
[9]Error parsing value '475000' for field 'Used Length [mm]': dict values object is not subscriptable
[9]Error parsing value '475000' for field 'Calculated Length [mm]': dict values object is not subscriptable
[... cut ...]
[84] Mandatory value for column 'Start Datetime [dd.mm.yyyy hh:mm]' is missing!
[84] Mandatory value for column 'Duration' is missing!
[84] Mandatory value for column 'File Name' is missing!

[octoprint.log](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/6527891/octoprint.log)
[printJobHistory.zip](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/6527894/printJobHistory.zip)
[OctoprintPrintJobHistory.050121.csv](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/6527906/OctoprintPrintJobHistory.050121.csv)

Thanks in advance

OllisGit, 2021-05-23:
Hi @santanig5,
I was a little bit confused, because the initial ticket was for "not showing items" and for the other issue there was the csv-file missing.
Anyway, I looked into your provided CSV  and there where a lot of empty lines like this (provided by excel):
```
,,,,,,,,,,,,,,,,,,,,
```
Also the filament-data import was broken in the plugin.

Both issues will be fixed in next release (next days)

santanig5, 2021-05-23:
Great!
I’ll be on the look out for the update.

santanig5, 2021-05-25:
Yep, new version allowed me to import the data, since I did not know the actual start date, I just used the end date, but at least the duration is correct.

Thanks!