OllisGit/OctoPrint-PrintJobHistory #140: Print History not visible after manual add of print

taker218, 2021-04-19:
So I just added a print via the "Add printjob" and now all the prints are invisible.
But when I go to statistics, I get all the statistics shown, including the new one I added (had 609 prints before, now it's 610).

[image]
[image]

Version is the latest 1.12.dev1.

Any idea why this is happening? Or is there any logging I can activate to see what's going on?

taker218, 2021-04-19:
Okay, I just clicked on one of the sortings and now the prints are there again, but it seems like I can't show the page with the manually added print.
All other pages show just fine, only the one with the new print doesn't show up.

taker218, 2021-04-19:
I attached my database, with just the last "normal" print and the one I added through the plugin. As well as the octoprint.log from the last startup.

the screenshot shows which information I provided.
[image]
[issue140_printjobhistory.zip](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/6336028/issue140_printjobhistory.zip)

OllisGit, 2021-05-16:
Hi,

new version 1.3.0 is out and should fix the issue.

Please test and give feedback.

Thx, in advance
Olli

taker218, 2021-05-18:
Just installed the new version and now I can add a new print and it shows up in the gui.

closed 2021-05-19 by OllisGit