OllisGit/OctoPrint-PrintJobHistory #141: Print Job Dialog does not appear after successfull print

realbadfox, 2021-04-20:
Since last update, the Dialog after a successfull print does not appear, and was not saved.

realbadfox, 2021-04-20:
[octoprint.log](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/6342221/octoprint.log)

zhechenzhu, 2021-04-21:
save error message here. 'SpoolmanagerPlugin' object has no attribute 'myFilamentOdometer'

adam5zki, 2021-04-21:
I have the same issue. same error in the octoprint.log.

EvilBlood, 2021-04-22:
Same over here

chimcen, 2021-04-23:
Here the same - the fix from older version doesn't work anymore...

realbadfox, 2021-04-23:
installed last working Release V1.10.0, with that it is working

speedysk1, 2021-04-26:
Add me to the list of the plugin no longer working.  :(

melvinmicallef, 2021-04-27:
not working.. i dont see it worth it to switch to filament manager from spool manager just cause of this bug..so sadly i will have to uninstall this plugin

Filmman586, 2021-05-05:
I'm having the same issue. It use to work great but now it doesn't. After a successful print, nothing happens. I used to get the edit dialogue window, now nothing, and now I have to refresh my window for the print to show up.

mediamanrit, 2021-05-11:
I'll add a me-too here.  No tracebacks in the octoprint log file though.  Still investigating.  It's not due to SpoolManager or FilamentManager conflicts in my case as I had neither installed when this started happening.

OllisGit, 2021-05-16:
Hi,

new version 1.3.0 is out and should fix the issue.

Please test and give feedback.

Thx, in advance
Olli

mediamanrit, 2021-05-20:
> Hi,
> 
> new version 1.3.0 is out and should fix the issue.
> 
> Please test and give feedback.
> 
> Thx, in advance
> Olli

Seems to be working now consistently.  Thanks!

chimcen, 2021-05-20:
This was working since 1.13.2 in combination with Filament Manager again…

closed 2021-05-22 by OllisGit