OllisGit/OctoPrint-PrintJobHistory #142: Unable to manually add snapshots to print job after job finishes

clarkiej, 2021-04-26:
Using PrintJobHistory 1.12.dev1. I have Filament Manager 1.72 and SpoolManager 1.3.4 with FilamentManager set as the tracker. I was able to manually load snapshots after a print job using version 1.70 but this is no longer the case.

Spool, Spool Vendor and material also are blank under the General tab
[image]
 but the spool and material show up under the ToolMeasurement tab
[image]

gbomacfly, 2021-05-01:
Same here:
`2021-05-01 10:04:21,257 - octoprint.plugins.PrintJobHistory.DatabaseManager.SQL - DEBUG - ('SELECT "t1"."databaseId", "t1"."created", "t1"."userName", "t1"."fileOrigin", "t1"."fileName", "t1"."filePathName", "t1"."fileSize", "t1"."printStartDateTime", "t1"."printEndDateTime", "t1"."duration", "t1"."printStatusResult", "t1"."noteText", "t1"."noteDeltaFormat", "t1"."noteHtml", "t1"."printedLayers", "t1"."printedHeight", "t1"."slicerSettingsAsText" FROM "pjh_printjobmodel" AS "t1" WHERE ("t1"."databaseId" = ?) LIMIT ? OFFSET ?', [114, 1, 0])
2021-05-01 10:04:21,275 - octoprint.plugins.PrintJobHistory.DatabaseManager.SQL - DEBUG - ('SELECT "t1"."databaseId", "t1"."created", "t1"."printJob_id", "t1"."profileVendor", "t1"."diameter", "t1"."density", "t1"."material", "t1"."spoolName", "t1"."spoolCost", "t1"."spoolCostUnit", "t1"."spoolWeight", "t1"."usedLength", "t1"."calculatedLength", "t1"."usedWeight", "t1"."usedCost", "t1"."toolId" FROM "pjh_filamentmodel" AS "t1" WHERE ("t1"."printJob_id" = ?)', [114])                                                                  
2021-05-01 10:04:21,281 - octoprint - ERROR - Exception on /plugin/PrintJobHistory/storePrintJob/114 [PUT]                                                                        
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app                                                                                         
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request                                                                            
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception                                                                            
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise                                                                                        
    raise value
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request                                                                            
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request                                                                                 
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1941, in wrapper                                                                            
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/api/PrintJobHistoryAPI.py", line 304, in put_printjob                                               
    self._updatePrintJobFromJson(printJobModel, jsonData)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/api/PrintJobHistoryAPI.py", line 63, in _updatePrintJobFromJson                                     
    filamentModel.profileVendor = self._getValueFromJSONOrNone("spoolVendor", jsonData)
AttributeError: 'list' object has no attribute 'profileVendor'`

OllisGit, 2021-05-16:
Hi @clarkiej, @gbomacfly ,

new version 1.3.0 is out and should fix the issue.

Please test and give feedback.

Thx, in advance
Olli

clarkiej, 2021-05-17:
Thank you for the info. I have downloaded the update and can now manually attach images to the print job again. much appreciated.

clarkiej, 2021-05-17:
I have downloaded the update and can manually attach images to the print job again.

 

Much appreciated.

OllisGit, 2021-05-17:
Hi @clarkiej ,
thanks for the fast response!

BR
Olli

closed 2021-05-17 by OllisGit