OllisGit/OctoPrint-PrintJobHistory #143: Plugin throws exception: no job logging

apauli01, 2021-05-05:
Hey, I installed the plugin and unfortunately it does not work.
Here is the log:
```
2021-05-05 11:23:39,359 - octoprint - ERROR - Exception on /plugin/PrintJobHistory/loadStatisticByQuery [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/_compat.py", line 39, in reraise
    raise value
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1941, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/api/PrintJobHistoryAPI.py", line 181, in get_statisticByQuery
    statistic = self._databaseManager.calculatePrintJobsStatisticByQuery(tableQuery)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/DatabaseManager.py", line 469, in calculatePrintJobsStatisticByQuery
    fromToString = firstDate.strftime('%d.%m.%Y %H:%M') + " - " + lastDate.strftime('%d.%m.%Y %H:%M')
AttributeError: 'NoneType' object has no attribute 'strftime'
2021-05-05 11:23:51,707 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.1.20
```

donlook, 2021-05-07:
I am having the same issue as well.
Jobs are not recorded:
OctoPrint 1.6.0
PrintJobHistory 1.12.dev1
SpoolManager 1.3.4

2021-05-07 16:22:33,574 - octoprint.plugin - ERROR - Error while calling plugin PrintJobHistory
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1941, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 818, in on_event
    self._printJobFinished("success", payload)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 627, in _printJobFinished
    self._createAndAssignFilamentModel(self._currentPrintJobModel, payload)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 226, in _createAndAssignFilamentModel
    filamentExtrusionArray = self._readMeasuredFilament()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 402, in _readMeasuredFilament
    result = self._spoolManagerPluginImplementation.myFilamentOdometer.getExtrusionAmount()
AttributeError: 'SpoolmanagerPlugin' object has no attribute 'myFilamentOdometer'

apauli01, 2021-05-07:
Found out the SpoolManager causes crashs. Using FillamentManager instead solves the problem.

OllisGit, 2021-05-16:
Hi,

new version 1.3.0 is out and fix some issues.
HINT: Still not working with current SpoolManager V1.3.4 you need to wait for V1.4.0, sorry!

Please test and give feedback.

@apauli01 could you provide me your database, I think the enddate is not filled correctly. You can download the database in the plugin-settings. See storage-tab.

Thx, in advance
Olli

donlook, 2021-05-16:
Hello Olli,
Here is my database file.
Regards,
Don

[OctoprintPrintJobHistory.csv](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/6489720/OctoprintPrintJobHistory.csv)

OllisGit, 2021-05-16:
Hi @donlook,
thanks for the fast response. Please also attach the binary database and not the csv.
[image]

donlook, 2021-05-16:
Hello Olli,

Here you go. 

My current version info:
OctoPrint 1.6.1
PrintJobHistory 1.13.0
SpoolManager 1.3.4

I have not printed any jobs since upgrading to 1.13.0.
Let me know if there is any additional information that I can provide.

Don

[printJobHistory.zip](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/6489801/printJobHistory.zip)

OllisGit, 2021-05-17:
fyi @donlook your database is very helpfull. I found some issues that are related to manually created jobs. I need to analyse a little bit deeper, take some time!

donlook, 2021-05-17:
Olli,

You may have already figured this out, but if I select a timeframe of the last 7 days, the jobs display correctly, sort and filter with the options works as well.

Don

OllisGit, 2021-05-18:
Hi @apauli01  , @donlook  ,

new Version 1.13.1 is out and should fix the issue.

Please test and give feedback.

Thx, in advance
Olli

donlook, 2021-05-18:
Hello @OllisGit ,

I installed the new version and all of the jobs in my database are now visible and the filters and sort options work great.

I also uploaded some new print jobs with the csv import feature. This worked as well. I can see the new jobs.

Great work,
Don

closed 2021-05-19 by OllisGit