OllisGit/OctoPrint-PrintJobHistory #145: Error, No logging since update 1.13.0 with Filament Manager

Stefan-deHoDev, 2021-05-17:
Since updating to version 1.13.0 automatic logging has stopped on both of my printers.

Looking at the log, here is the stacktrace:
```

2021-05-17 15:14:12,533 - octoprint.plugins.PrintJobHistory - INFO - Print result:success, CaptureMode:always
2021-05-17 15:14:12,533 - octoprint.plugins.PrintJobHistory - INFO - Start capturing print job
2021-05-17 15:14:12,552 - octoprint.plugins.PrintJobHistory.SlicerSettingsParser - INFO - Start parsing Slicer-Settings
2021-05-17 15:14:12,651 - octoprint.util.comm - INFO - Changing monitoring state from "Finishing" to "Operational"
2021-05-17 15:14:16,045 - octoprint.plugins.PrintJobHistory.SlicerSettingsParser - INFO - Finished parsing Slicer-Settings
2021-05-17 15:14:16,051 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Thumbnail is present
2021-05-17 15:14:16,063 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Try converting thumbnail '/home/pi/.octoprint/data/PrintJobHistory/../UltimakerFormatPackage/AI3M_Anycubic_Spool_Abstandshalter.png' to '/home/pi/.octoprint/data/PrintJobHistory/snapshots/20210517-142817.jpg'
2021-05-17 15:14:17,608 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Converting successfull!
2021-05-17 15:14:17,625 - octoprint.plugins.PrintJobHistory - INFO - There is no calculated filament data in meta-file?
2021-05-17 15:14:17,672 - octoprint.plugin - ERROR - Error while calling plugin PrintJobHistory
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1941, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 757, in on_event
    self._printJobFinished("success", payload)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 565, in _printJobFinished
    self._createAndAssignFilamentModel(self._currentPrintJobModel, payload)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 305, in _createAndAssignFilamentModel
    allFilamentsWithoutTotal = printJob.getFilamentModels(withoutTotal=True)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/models/PrintJobModel.py", line 62, in getFilamentModels
    allFilamentModels[:] = [filamentModel for filamentModel in allFilamentModels if filamentModel.toolId != "total"]
TypeError: 'dict_values' object does not support item assignment
```

It worked before the update.

However even before the update the fields "Spool", "Spool Vendor" and "Material" were not filled by Filament Manager. Though in the "Tool measurement" tab "Spool" and "Material" were both correctly filled with data from Filament Manager.

I hope this gets resolved quickly

ivgh1992, 2021-05-17:
Same problem here. It stopped working since the last update. It doesn't even register the new print jobs since May the 17th, 2021. Hopefully this problem gets solved, it used to work great

ivgh1992, 2021-05-17:
`2021-05-17 13:37:26,170 - octoprint.plugins.softwareupdate - INFO - Starting update of PrintJobHistory to 1.13.0...
2021-05-17 13:37:38,331 - octoprint.plugins.softwareupdate - INFO - Update of PrintJobHistory to 1.13.0 successful!
2021-05-17 13:37:38,435 - octoprint.plugins.softwareupdate - INFO - Starting update of firmwareupdater to 1.11.0...
2021-05-17 13:37:44,205 - octoprint.plugins.softwareupdate - INFO - Update of firmwareupdater to 1.11.0 successful!
2021-05-17 13:37:44,372 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2021-05-17 13:37:44,374 - octoprint.plugins.softwareupdate - INFO - Restarting...
2021-05-17 13:37:53,870 - octoprint.plugins.PrintJobHistory - INFO - Start initializing
2021-05-17 13:37:53,870 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Init DatabaseManager
2021-05-17 13:37:53,871 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Using database in: /home/pi/.octoprint/data/PrintJobHistory/printJobHistory.db
2021-05-17 13:37:53,874 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Check if database-scheme upgrade needed.
2021-05-17 13:37:53,885 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - We need to upgrade the database scheme from: '4' to: '5'
2021-05-17 13:37:53,892 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Backup of printjobhistory database created '/home/pi/.octoprint/data/PrintJobHistory/printJobHistory-backup-V4-20210517-1337.db'
2021-05-17 13:37:53,892 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Database migration from '4' to '5'
2021-05-17 13:37:53,892 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO -  Starting 4 -> 5
2021-05-17 13:37:53,924 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO -  Successfully 4 -> 5
2021-05-17 13:37:53,925 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Database-scheme successfully upgraded.
2021-05-17 13:37:53,925 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Done DatabaseManager.createDatabase
2021-05-17 13:37:53,925 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Init CameraManager
2021-05-17 13:37:53,925 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Snapshot-Folder:/home/pi/.octoprint/data/PrintJobHistory/snapshots
2021-05-17 13:37:53,926 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Done CameraMenager
2021-05-17 13:37:53,928 - octoprint.plugins.PrintJobHistory - INFO - Done initializing
2021-05-17 14:28:17,812 - octoprint.plugins.PrintJobHistory - INFO - Print result:success, CaptureMode:always
2021-05-17 14:28:17,812 - octoprint.plugins.PrintJobHistory - INFO - Start capturing print job
2021-05-17 14:28:17,823 - octoprint.plugins.PrintJobHistory.SlicerSettingsParser - INFO - Start parsing Slicer-Settings
2021-05-17 14:28:17,880 - octoprint.util.comm - INFO - Changing monitoring state from "Finishing" to "Operational"
2021-05-17 14:28:18,070 - octoprint.plugins.PrintJobHistory.SlicerSettingsParser - INFO - Finished parsing Slicer-Settings
2021-05-17 14:28:18,073 - octoprint.plugins.PrintJobHistory - WARNING - Thumbnail not found in print metadata
2021-05-17 14:28:18,075 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Try taking snapshot '/home/pi/.octoprint/data/PrintJobHistory/snapshots/20210517-134917.jpg' from 'http://127.0.0.1:8080/?action=snapshot'
2021-05-17 14:28:18,099 - octoprint.plugin - ERROR - Error while calling plugin PrintJobHistory`

ivgh1992, 2021-05-17:
2021-05-17 13:37:26,170 - octoprint.plugins.softwareupdate - INFO - Starting update of PrintJobHistory to 1.13.0...
2021-05-17 13:37:38,331 - octoprint.plugins.softwareupdate - INFO - Update of PrintJobHistory to 1.13.0 successful!
2021-05-17 13:37:38,435 - octoprint.plugins.softwareupdate - INFO - Starting update of firmwareupdater to 1.11.0...
2021-05-17 13:37:44,205 - octoprint.plugins.softwareupdate - INFO - Update of firmwareupdater to 1.11.0 successful!
2021-05-17 13:37:44,372 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2021-05-17 13:37:44,374 - octoprint.plugins.softwareupdate - INFO - Restarting...
2021-05-17 13:37:53,870 - octoprint.plugins.PrintJobHistory - INFO - Start initializing
2021-05-17 13:37:53,870 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Init DatabaseManager
2021-05-17 13:37:53,871 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Using database in: /home/pi/.octoprint/data/PrintJobHistory/printJobHistory.db
2021-05-17 13:37:53,874 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Check if database-scheme upgrade needed.
2021-05-17 13:37:53,885 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - We need to upgrade the database scheme from: '4' to: '5'
2021-05-17 13:37:53,892 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Backup of printjobhistory database created '/home/pi/.octoprint/data/PrintJobHistory/printJobHistory-backup-V4-20210517-1337.db'
2021-05-17 13:37:53,892 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Database migration from '4' to '5'
2021-05-17 13:37:53,892 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO -  Starting 4 -> 5
2021-05-17 13:37:53,924 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO -  Successfully 4 -> 5
2021-05-17 13:37:53,925 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Database-scheme successfully upgraded.
2021-05-17 13:37:53,925 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Done DatabaseManager.createDatabase
2021-05-17 13:37:53,925 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Init CameraManager
2021-05-17 13:37:53,925 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Snapshot-Folder:/home/pi/.octoprint/data/PrintJobHistory/snapshots
2021-05-17 13:37:53,926 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Done CameraMenager
2021-05-17 13:37:53,928 - octoprint.plugins.PrintJobHistory - INFO - Done initializing
2021-05-17 14:28:17,812 - octoprint.plugins.PrintJobHistory - INFO - Print result:success, CaptureMode:always
2021-05-17 14:28:17,812 - octoprint.plugins.PrintJobHistory - INFO - Start capturing print job
2021-05-17 14:28:17,823 - octoprint.plugins.PrintJobHistory.SlicerSettingsParser - INFO - Start parsing Slicer-Settings
2021-05-17 14:28:17,880 - octoprint.util.comm - INFO - Changing monitoring state from "Finishing" to "Operational"
2021-05-17 14:28:18,070 - octoprint.plugins.PrintJobHistory.SlicerSettingsParser - INFO - Finished parsing Slicer-Settings
2021-05-17 14:28:18,073 - octoprint.plugins.PrintJobHistory - WARNING - Thumbnail not found in print metadata
2021-05-17 14:28:18,075 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Try taking snapshot '/home/pi/.octoprint/data/PrintJobHistory/snapshots/20210517-134917.jpg' from 'http://127.0.0.1:8080/?action=snapshot'
2021-05-17 14:28:18,099 - octoprint.plugin - ERROR - Error while calling plugin PrintJobHistory

OllisGit, 2021-05-18:
fyi: I think I found the issue. It is related to Python 3.7. Hopefully I can fix this in the next days.

OllisGit, 2021-05-18:
Hi @Stefan-deHoDev , @ivgh1992 ,

new Version 1.13.1 is out and should fix the issue.

Please test and give feedback.

Thx, in advance
Olli

ivgh1992, 2021-05-18:
Hi @OllisGit ,
the new version is logging the print jobs again, however there's still a problem with the csv-file export.  The Spool data (Name, material, vendor,etc.) is not being filled in the csv file.
[image]

Stefan-deHoDev, 2021-05-19:
Version 1.13.1 works again, thank you for your quick response!!

@OllisGit
This issue can be closed.

To elaborate on what @ivgh1992 is saying. Version 1.13.1 logs spool data correctly but exporting to CSV leaves those fields empty.

Seems to be a problem with the exporter not the logging. 

Probably better to create a new issue for this.

OllisGit, 2021-05-19:
Hi @ivgh1992,
I fixed the export in the new release 1.13.2
Please test and give feedback.

As @Stefan-deHoDev said, it was a different issue, so next time would be nice if you could create a new issue 
(then I don't need to do that ;-). 

Thx, for testing
Olli

ivgh1992, 2021-05-19:
HI @OllisGit,
Thank you for your response. It works perfectly now!

closed 2021-05-22 by OllisGit