OllisGit/OctoPrint-PrintJobHistory #146: CSV import/reimport on v1.13.2

chimcen, 2021-05-20:
Hi,

after actualizing to latest PJHv1.13.2 i've tryed to reimport my old csv export and get only some errors.

I thought, there could be some changes within the csv layout, so i've deleted the database and added a sample print within PJH manually.
After then i've exported this to csv, deleted the database again and tried to reimport the prev. export - ending also in error messages:

[image]

Can you help?

chimcen, 2021-05-20:
Replacing the not parsable values (except used weight), the import is possible:
[image]

OllisGit, 2021-05-24:
Hi,
in the latest release V1.14.0 the issue should be fixed.

Please test and give me a feedback.

Thx, in advance
Olli

chimcen, 2021-05-26:
Hi Olli,

CSV Import works now.
But there is a problem with Spool Manager and PHJ togehter - i'll post this in Spool Manager git...

closed 2021-05-26 by chimcen