OllisGit/OctoPrint-PrintJobHistory #147: No dialog or recording of print jobs

speedysk1, 2021-05-21:
I have been waiting for the fix in 1.13, but even after upgrading to 1.13.2, I'm still no longer getting the print history dialog and it's not saving the history either.  Last print job shows date of 4/19/2021 and I print on a near daily basis.  Not sure what logs you might need, but happy to provide.

chimcen, 2021-05-21:
Did you hav installed all depencies?
For me it seems it was needed to install Spoolmanager - although it doesn‘t work with PJH atm -,too.
I‘ve installed all depecies and get the filament data atm from Filament Manager plugin…

…but i‘ve also cleaned my database before using 1.13.2…but this isn‘t a good solution if it‘s the solution, because reimport saved csv export doesn‘t work… :(

cjmalchow, 2021-05-22:
I'm having the same problem. Since updating to the latest version of the plugin, a few days ago, it is no longer recording any history.

OllisGit, 2021-05-22:
Hi,

please provide me the octoprint.log and also the printjobHistory.db
[image]

cjmalchow, 2021-05-23:
For sure! Attached are the files you've requested. If I can help further
please let me know! I really like your plugin and would like it even better
if it was working again ;)

Cameron

OllisGit, 2021-05-23:
Hi @cjmalchow,

answering via EMail and attaching a file is not possible with GitHub.
You need to upload the file via Web-Browser!

cjmalchow, 2021-05-24:
Find attached the requested files.

[octoprint-logs.zip](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/6530630/octoprint-logs.zip)
[printJobHistory.zip](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/6530635/printJobHistory.zip)

RChicago, 2021-05-24:
I am having the same issue....waiting of how to fix!

OllisGit, 2021-05-24:
Hi,

the SpoolManager-Version you used was not compatible with the PJH-Plugin.
Log-Message:
```
'SpoolmanagerPlugin' object has no attribute 'myFilamentOdometer'
```

Try to use the lastest PJH-Plugin V1.14.0 and SpoolManger V1.4.0, that should work!

Please test and give feedback!
Olli

chimcen, 2021-05-26:
Had the same problem with SpoolManage and PJH also.
After updating to PJH 1.14.0 and SpoolManager 1.4.3 - this problem is solved for me.

Btw.: Olli, is it neccessary to have Filamentmanger also installed when using SpoolManager?

OllisGit, 2021-05-27:
@chimcen it is not necessary to install both. You can decide which one you want to use.

chimcen, 2021-05-28:
Thats what i thought already - just the dialog "...to get the real power of PJH install FilamentManager..." if dependency check is switched on and SpoolManager is active confusing a little bit...

RChicago, 2021-05-28:
Is one better than the other??

OllisGit, 2021-10-30:
@RChicago .."better", you need to decide. 
FM can work with external database, so it can e used with multiple OP-Instances.
SPM has more features and development is ongoing.

closed 2021-10-30 by OllisGit