OllisGit/OctoPrint-PrintJobHistory #148: AttributeError: 'SpoolModel' object has no attribute 'weight'

Julaho, 2021-05-25:
Even with both plugins, PrintJobHistory and SpoolManager, updated to the latest versions my printjobs are not added to the history. The log shows the folloing error:

```
2021-05-25 10:50:24,559 - octoprint.plugin - ERROR - Error while calling plugin PrintJobHistory
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1941, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 850, in on_event
    self._printJobFinished("success", payload)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 649, in _printJobFinished
    self._createAndAssignFilamentModel(self._currentPrintJobModel, payload)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 276, in _createAndAssignFilamentModel
    selectedSpoolDataDict = self._getSelectedSpools()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintJobHistory/__init__.py", line 432, in _getSelectedSpools
    selectedSpoolInformations = self._spoolManagerPluginImplementation.api_getSelectedSpoolInformations()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_SpoolManager/__init__.py", line 480, in api_getSelectedSpoolInformations
    "weight": spoolModel.weight
AttributeError: 'SpoolModel' object has no attribute 'weight'
```

OllisGit, 2021-05-25:
Hi @Julaho,

please update to SpoolManager V1.4.2, this should solve the issue.
Please test and give feedback.

Thx, in advance
Olli

Julaho, 2021-05-25:
Hi @OllisGit,

I just updated to V1.4.2 and made a quick test-print.
The Plugin ist working again - thanks for your great support! Enjoy your coffee. ;-)

Best, Julian

OllisGit, 2021-05-25:
Hi @Julaho,

thanks for testing and the coffee!!!

BR
Olli

closed 2021-05-25 by OllisGit