OllisGit/OctoPrint-PrintJobHistory #149: disabling of filamenttracking is now possible → but why is it auto-disabled?

codingcatgirl, 2021-05-25:
After updating to the latest version i got a message telling me to select a plugin for filament tracking. Apparently you added the feature to *disable* it in the last version. However, i feel like it shouldn't have auto-disabled itself with that update, right?

OllisGit, 2021-05-27:
Hi @codingcatgirl!

Well, the older version always selected the filamenttracker-plugin. If both installed, it selected Filamenttracker.
Now the SpoolManager is more stable, I want that the user is explicit selecting the prefered Plugin.

Now it is "auto-disabled" after you disabled the plugin and after reenabling it you must select a plugin.

There is also an other feature in my roadmap: collecting filament-infos from gcode instead of tracking plugin.

But it shouldn't be problem for the user to select a tracker, right?

codingcatgirl, 2021-05-27:
Just easy to misunderstand, i was confused because it suddenly wanted me to do that :)

donlook, 2021-06-01:
I have also noticed that the plugin selection seems to be forgotten when OctoPrint is restarted. FilamentManager is the only plugin that I have installed and I noticed the message during the restart that I should choose a plugin.

It does not seem to happen on every restart, but if I restart several times in a row, it will forget the selection.

The attached log file has a number of restarts where the setting was both remembered and forgotten.

[octoprint.zip](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/6577757/octoprint.zip)

dualznz, 2021-06-03:
1

> I have also noticed that the plugin selection seems to be forgotten when OctoPrint is restarted. FilamentManager is the only plugin that I have installed and I noticed the message during the restart [...]

I have this same issue on auto disabled when pi is restarted

ManuelMcLure, 2021-06-13:
This appears to be a race condition - the code to check for the filament manager is called before the Filament Manager plugin is completely initialized so it shows up as disabled and the setting is set to None.

ManuelMcLure, 2021-06-13:
I've entered a PR to fix this. If you want to test whether it fixes it for you, install the version from https://github.com/ManuelMcLure/OctoPrint-PrintJobHistory/archive/refs/heads/fixfmreset.zip

donlook, 2021-06-13:
@ManuelMcLure 

I installed version 1.14.1,

When I restarted OctoPrint and looked at the filament management selection in the Print Job History plugin, both radio buttons for SpoolManger and FilamentManager are disabled. I have FilamementManager installed. SpoolManager is not installed in my configuration. See the attached screen captures. It does not seem to forget the assignment.

Thanks, let me know if I can supply any more information.

[image]

[image]

ManuelMcLure, 2021-06-13:
@donlook did you try installing from the URL I provided using the "Get More->URL" option? The default 1.14.1 that gets installed normally does not have my fix yet.

donlook, 2021-06-13:
@ManuelMcLure 
I installed from here:
https://github.com/ManuelMcLure/OctoPrint-PrintJobHistory/archive/refs/heads/fixfmreset.zip

ManuelMcLure, 2021-06-13:
@donlook I can't reproduce it - I have the same setup as you (Filament Manager installed, Spool Manager not installed) and I see the Filament Manager radio button not disabled.

ManuelMcLure, 2021-06-13:
This is probably a timing issue, which can be difficult to reproduce and fix.

ManuelMcLure, 2021-06-13:
@donlook Can you try forcing a reload with Ctrl+Shift+R and see if that fixes it?

ManuelMcLure, 2021-06-13:
@donlook could you try uninstalling and reinstalling the plugin from the same URL? I just pushed another fix that _may_ fix your issue.

donlook, 2021-06-13:
@ManuelMcLure 
Before I installed the new fix that you provided, I did a system reboot. After that, I could not reproduce the problem. What I was seeing then that was different was the "Settings Update Detected" dialog when opening the OctoPrint Settings dialog.

After that testing, I installed your newest version. The newest version does not have the "settings changed" dialog. The original problem that I saw does not appear. I have restarted OctoPrint several times and also restarted the system as well. I have also tried it with my printer powered off as well as powered on. 

The problem seems to be resolved.

[image]

ManuelMcLure, 2021-06-13:
Thanks for testing. Hopefully the PR gets merged soon.

codingcatgirl, 2021-06-30:
Restarted. Removed the setting again. Hm.

codingcatgirl, 2021-06-30:
(i'm still just running the latest version)

MrSco, 2021-10-30:
Any updates on this? I set the spoolmanager filament tracker in the settings but every time I reboot, the PJHM notification shows.

Nazar78, 2021-11-07:
Getting the same issue, everytime the server reboot I also see this in the log:

2021-11-08 00:31:55,224 - octoprint.plugins.PrintJobHistory - ERROR - Filamentracking is disabled, but some plugins are installed!

OllisGit, 2021-11-13:
Hi,
new release [1.15.0](https://github.com/OllisGit/OctoPrint-PrintJobHistory/releases/tag/1.15.0) is available and fix this issue.

Please test and give me a feedback. 
Thx, in advance!
Olli

closed 2022-02-28 by github-actions[bot]