OllisGit/OctoPrint-PrintJobHistory #150: [Feature Request]

chimcen, 2021-05-26:
Is it possible to change the behavior how the screenshot filename will be created?

Now the name will be matched to the start date and time of the print - if its possible to change it to the name of the gcode file, its possible to safe filespace and loading times of PJH plugin frame...

e.g.
When i'm printing the same model on two different dates, two identical screenshots will be stored and also loaded on opening PJH plugin.
If the screenshot file name could be changed to the gcode filename it would be only one screenshot saved and also loaded on opening PJH plugin for multiple prints of the model - no matter when they are printed...

OllisGit, 2021-05-27:
@chimcen , but what if you want to print the same gcode and exchange the screenshot only in one job. Then in both jobs the srcreenshot is changed. 

Maybe I can think about a feature like "linked screenshots"...I also want to support to store multiple images (e.g. https://plugins.octoprint.org/plugins/multicam/)...it's on my list.

chimcen, 2021-05-28:
Why i should change in this case the screenshot for one print if they are exactly the same print from the same gcode?

Good to see the roadmap - muliple images (from gcode + cams) is a cool idea...

closed 2022-02-02 by github-actions[bot]