OllisGit/OctoPrint-PrintJobHistory #151: Statistic for PrintJobHistory

beunoutch, 2021-05-26:
Hello This request concerns the improvement of the statistics of Octoprint-PrintJobHistory.
As the sqlite database is made up of all impressions, is it possible to add to this statistic a bit.

It would be great to have an interface on which you could choose the statistic you want, embellished with a graph.

examples of interesting stats with graphics
Number of impressions per year, per month, per day.
Printing times per year, per month, per day
Number of completed / failed prints.
cumulative cost estimates ...
Statistics of coils (combined with filament manager), the most used ...

...
...

OllisGit, 2021-05-27:
Hi @beunoutch,
statistics/reports is an endless topic. Each user wants different data in a different format.
That's the reason why I decided to  just make a simple stats-popup instead of a full blown stats-plugin, like https://plugins.octoprint.org/plugins/stats/

But my goal is to "open" all the PrintJob-Data to other plugins, via WebService, WebSocket or Plugin-APIs.
And I also want to store the data into a "real" database. So, that the user can connect to the database and generate it's own report.
Maybe you can ask in the the stats-plugin author about an integration or ask in the OP-Forum (https://community.octoprint.org/) about developing such plugin.

beunoutch, 2021-06-02:
I thought about it to what will be important to have in statistics it would just be necessary:
by years:
- the number of prints per year.
- the weight used per year.

this will provide an overall summary of the use of the printer.
can you do this to us?

OllisGit, 2021-06-13:
Hi @beunoutch,
what you request is already there. Just define the time-range you like to see, select all jobs and click on "Statistics".
[image]

closed 2021-10-30 by OllisGit