OllisGit/OctoPrint-PrintJobHistory #152: Last update broke Print Time Genius...

cjmalchow, 2021-05-27:
After updating to recent version 1.14.0 the estimated print times and reported percentage through Dashboard and reported via DisplayLayerProgress were wildly off. Before even printing Octoprint would report the print would be at least 85% complete, along with the GCode, and move up from there, generally saying the print will be done in the next couple minutes, despite it really being close to an hour, or more. I did check Print Time Genius to see what time it was calculating and it appears to be calculating properly, and when I disabled PrintJobHistory estimated times went back to making sense. Would love to be able to use plugin, but for now I can't :(

OllisGit, 2021-06-13:
hmmm..PJH-Plugin doesn't change the behaviour of estimation/percentage calculation, it even doesn't read these values . 
The plugin just capture the real start- and end-timestamp and calculate the duration between the two values.

On the other hand, PTG replace totally the build in OP percentage calculation.

I have no clue how PJH has an impact on that.
What happen if you just disable PTG, is it still broken?

OllisGit, 2021-10-30:
No feedback, raise a new ticket if needed.

closed 2021-10-30 by OllisGit