OllisGit/OctoPrint-PrintJobHistory #153: PJH skips logging the material from certain jobs

ivgh1992, 2021-05-27:
Hi Olli, thanks for the great plugin. I work with PJH and filament manager with a postgres database. I noted that sometimes PJH skips some jobs and doesn't log the material properties from those jobs. Even though the material is substracted from filament manager, the used material in PJH is logged as 0 and the material, spool, vendor etc stays blank. 
Some screenshots are attached and the octoprint log.
Thank you for your time! 
[octoprintLog.log](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/6556914/octoprintLog.log)

[image]
[image]
[image]

Stefan-deHoDev, 2021-05-28:
I'm having the same problem I think. Though I'm not sure, since it appears as if no print job has been logging Filament Data since I updated to the newest version. 

Yes, I selected Filament Manager after updating on both of my printers.

I looked at the log and there is no exception or error message
`
2021-05-28 13:22:33,818 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Finishing"
2021-05-28 13:22:33,829 - octoprint.filemanager.analysis - INFO - Starting analysis of local:BB1ABL_MediumWave02.gcode
2021-05-28 13:22:33,833 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python3 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 --bed-z=0.0 /home/pi/.octoprint/uploads/BB1ABL_MediumWave02.gcode
2021-05-28 13:22:33,879 - octoprint.printer.standard.job - INFO - Print job done - origin: local, path: BB1ABL_MediumWave02.gcode, owner: stefan
2021-05-28 13:22:33,973 - octoprint.plugins.PrintJobHistory - INFO - Print result:success, CaptureMode:always
2021-05-28 13:22:33,974 - octoprint.plugins.PrintJobHistory - INFO - Start capturing print job
2021-05-28 13:22:33,995 - octoprint.plugins.PrintJobHistory.SlicerSettingsParser - INFO - Start parsing Slicer-Settings
2021-05-28 13:22:34,052 - octoprint.util.comm - INFO - Changing monitoring state from "Finishing" to "Operational"
2021-05-28 13:22:34,229 - octoprint.plugins.tracking - INFO - Sent tracking event print_done, payload: {'origin': 'local', 'file': '3365a1168958b1debddfb4eebefd6842d8f6b3b6', 'elapsed': 14298}
2021-05-28 13:22:36,067 - octoprint.plugins.PrintJobHistory.SlicerSettingsParser - INFO - Finished parsing Slicer-Settings
2021-05-28 13:22:36,121 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Thumbnail is present
2021-05-28 13:22:36,311 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Try converting thumbnail '/home/pi/.octoprint/data/PrintJobHistory/../UltimakerFormatPackage/BB1ABL_MediumWave02.png' to '/home/pi/.octoprint/data/PrintJobHistory/snapshots/20210528-092415.jpg'
2021-05-28 13:22:36,821 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Converting successfull!
2021-05-28 13:22:37,061 - octoprint.plugins.PrintJobHistory - INFO - There is no calculated filament data in meta-file?
2021-05-28 13:22:39,807 - octoprint.plugins.excluderegion - INFO - Printing stopped: event=PrintDone
2021-05-28 13:22:39,808 - octoprint.plugins.filamentmanager - INFO - Updating Filament usage for octoprint configured toolcount: 1
2021-05-28 13:22:39,808 - octoprint.plugins.filamentmanager - INFO - Filament tracked toolcount: 1
2021-05-28 13:22:39,809 - octoprint.plugins.filamentmanager - INFO - Filament tracked values: [12536.832240000818]
2021-05-28 13:22:39,809 - octoprint.plugins.filamentmanager - INFO - Filament used: 12536.832240000818 mm (tool0)
2021-05-28 13:22:39,868 - octoprint.plugins.filamentmanager - INFO - Updated remaining filament on spool 'Glitzer Ultramarinblau - Glitzer PLA (Redline)' from 982.942g to 945.5502769512898g (-37.39172304871022g)
`

OllisGit, 2021-05-30:
Hi @ikopytko,  @Stefan-deHoDev,

thanks for the log file, because there is one important line:
```
There is no calculated filament data in meta-file?
```
This mean that OP is not calculating the length of the print-job.
Please take a look into the upload-folder, there is a ```.metadata.json``` file. This file list all printjobs with addition meta-data.
[image]

There is an update strategy of this file, maybe it is turned off or it is executed to late. Switch to ```always```
Look at: Plugin-Settings / Features /
[image]

ivgh1992, 2021-05-30:
Hi @OllisGit ,
Thank you for your response. I changed it and it works again with my two printers.
Thank you for your time

closed 2021-05-30 by ivgh1992