OllisGit/OctoPrint-PrintJobHistory #154: Printjob Report

ThorstenStr, 2021-05-30:
A nice feature would be to print a kind of status report from the plugin in A4 format (laser printer) that can be attached to the printed parts. Especially when you have created a lot of orders for customers / friends you have such a nice report on material, printing time, printing day and costs.

OllisGit, 2021-06-04:
Report could look like this:
(source @ThorstenStr )
[image]

OllisGit, 2021-06-04:
Internal:

Challenges I need to solve:
1. Html-Report needs to be a single browser page, so that the user could use the browser print function
2. Opening the new Browser-Page needs to request the report.html page

--> 2. could be handled via https://docs.octoprint.org/en/master/plugins/mixins.html#uiplugin
With this approach I can reuse the the whole OP-mechanism (knockout viewmodel, asset loading)
Examples: 
https://github.com/OctoPrint/OctoPrint/blob/1.4.2/src/octoprint/plugins/loginui/__init__.py
https://github.com/OctoPrint/OctoPrint/blob/1.3.10/src/octoprint/plugins/forcelogin/__init__.py

--> 2 or the delivery of the page could be done via https://flask.palletsprojects.com/en/2.0.x/api/#response-objects 
Set the correct mimetype and send a html-string to the frontend
The html string could be build via the jinja2 template engine. No Javascript stuff needed. Rendering is fully done on server side. CSS needs to be included into html-code
See /src/octoprint/plugins/discovery/__init__.py
```
        response = flask.make_response(
            flask.render_template(
                "discovery.xml.jinja2",
                friendlyName=self.get_instance_name(),
                manufacturer=vendor,
                manufacturerUrl=vendorUrl,
                modelName=modelName,
                modelDescription=self._settings.get(["model", "description"]),
                modelNumber=self._settings.get(["model", "number"]),
                modelUrl=self._settings.get(["model", "url"]),
                serialNumber=self._settings.get(["model", "serial"]),
                uuid=self.get_uuid(),
                presentationUrl=flask.url_for("index", _external=True),
            )
        )
        response.headers["Content-Type"] = "application/xml"
        return response

<?xml version="1.0"?>
<root xmlns="urn:schemas-upnp-org:device-1-0">
    <specVersion>
        <major>1</major>
        <minor>0</minor>
    </specVersion>
    <device>
        <deviceType>urn:schemas-upnp-org:device:Basic:1</deviceType>
        <friendlyName>{{ friendlyName }}</friendlyName>
        <manufacturer>{{ manufacturer }}</manufacturer>
        {% if manufacturerUrl %}<manufacturerURL>{{ manufacturerUrl }}</manufacturerURL>{% endif %}
        <modelName>{{ modelName }}</modelName>
        {% if modelDescription %}<modelDescription>{{ modelDescription }}</modelDescription>{% endif %}
        {% if modelNumber %}<modelNumber>{{ modelNumber }}</modelNumber>{% endif %}
        {% if modelUrl %}<modelURL>{{ modelUrl }}</modelURL>{% endif %}
        {% if serialNumber %}<serialNumber>{{ serialNumber }}</serialNumber>{% endif %}
        <UDN>uuid:{{ uuid }}</UDN>
        <serviceList>
        </serviceList>
        <presentationURL>{{ presentationUrl }}</presentationURL>
    </device>
</root>
```

ThorstenStr, 2021-06-05:
I also think that an HTML page would be the best solution. I can also do the translations into different languages ​​for this. Is there also a kind of HTML form designer that can be integrated in order to adapt the look of the printout?

OllisGit, 2021-06-13:
Okay, the implementation "break-thru" is done.

It is now (next release) possible to create your own PrtinJob-Reports for a single PrintJob.
A detailed description how to use it, could be found in the [wiki](https://github.com/OllisGit/OctoPrint-PrintJobHistory/wiki/HowToPrintJobReport) 
The default report (A4-Format) looks like is:
[image]
If someone (with more UI skills)  could design a better one, I would be more then happy to replace the default report.

ThorstenStr, 2021-06-13:
Hello, this is absolutely perfect. Thank you for the extension.
I will adjust the appearance of the report and will be happy to share the results.
Many Thanks

OllisGit, 2021-11-13:
Hi @ThorstenStr,
it took some time...but now a new release [1.15.0](https://github.com/OllisGit/OctoPrint-PrintJobHistory/releases/tag/1.15.0) is available.

Please test and give me a feedback. 
Thx, in advance!
Olli

ThorstenStr, 2021-11-22:
Hi @OllisGit 
I'm currently testing it and so far I think it's great. however, on my first tests, the diameter and density are not displayed. (See attachement). Is there also a possibility not to show the UsedCosts but the TotalCosts?
I have already tested several parameters but without success.,
[3D-Druckjob-Report.pdf](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/7583916/3D-Druckjob-Report.pdf)

jessemillar, 2022-01-05:
I just found this feature yesterday and love it! Are there any plans to enable report creation for multiple history items at a time? I sometimes get orders for multiple prints and it'd be great to have a one-page invoice I can package in instead of one page per item.

OllisGit, 2022-01-16:
Hi @ThorstenStr,
you can change in the default template the output of diameter/density like this. Replace total with tool0:
[image]

Regarding "costs", currently there is a bug in the Cost-Plugin that prevents updating the cost-data in PJH, so when this is fixed it could be included here.

OllisGit, 2022-01-16:
Hi @jessemillar,
including a "multi-report-template" for the Job-Table should be possible....maybe of one of the next releases.

OllisGit, 2022-02-02:
fyi: I implemented the multi-print-report. You can print all, by search query or single selected
Not pretty, but the default report looks like this:
[image]

jessemillar, 2022-02-10:
@OllisGit Fantastic! That looks good enough for me (and I assume it's customizable with the template files). Is this functionality released? Or in a dev branch?

OllisGit, 2022-02-14:
Hi @jessemillar, @ThorstenStr,

I release a new version https://github.com/OllisGit/OctoPrint-PrintJobHistory/releases/tag/1.17.0.dev1  in **DEV channel**.   

This version include the option to export multiple printjobs as a a print-report. The report template is not very pretty...so if you have a better looking template I more then happy to include this as a default template.

ThorstenStr, 2022-02-18:
Hello Olli,
I was able to test the function for multiple prints and it looks great so far. So far I haven't found any errors. I saw the cost calculation is done in 1.17.0.dev1. Unfortunately, I don't really understand the handling. is there already a parameter for totalCosts?
Thank you for your great work

OllisGit, 2022-03-18:
Hi @ThorstenStr
answers of your questions/remarks

1.) In the past the total cost calculation was done by the [Cost-Plugin](https://github.com/OllisGit/OctoPrint-CostEstimation) on the frontend-side and transferred to the backend. But sometimes the browser is not used, so the total cost was always zero in the database. Now I am just reading the settings for calculating the costs (e.g. "Electricity cost") from the Plugin (via python, not JS) and do the calculation in PJH.
So, you still need the Cost-Plugin installed.

2.) There are two separate layouts (single/multi). So the single-layout is not included into the mutli-layout. If you change the singleLayout this is not used when you create the report from the job table view. Only when you open a print-job and press "Report" at the bottom.
Fist I though the singleLayout needs to be included to the multi layout, but now you have the option to create a rich detailed report for a single print job and a multi-report with less information.

3.) Open Tab-Issue will be fixed in next release.

ThorstenStr, 2022-03-19:
Hi Olli,

good job. It all looks good, I will test it further.
Can you give me the parameter for the costs "Total"? I tried it in different variants, but I don't get a value displayed.
Example: 
<div><u><p align="right">Printing costs: {{ printJobModelAsJson["filamentModels"]["total"]["totalCost"] }}€</p></u></div >

Maybe it's an idea for noobs like me ;) to create an overview of the parameters and their content in the wiki.

Overall, it's a great solution, I especially like the possibility of different layouts.

ThorstenStr, 2022-04-26:
Hi Olli,
do you already have an update for me? Unfortunately I haven't gotten it yet. 
Thank you 
Thorsten

closed 2022-06-07 by github-actions[bot]