OllisGit/OctoPrint-PrintJobHistory #156: Activate Plugin dependency check reporting incorectly

elminster, 2021-06-14:
Activate Plugin dependency check is reporting filament manager is missing, even though I have now switched job print history to use spool manager (filament manager still installed but now disabled).

OllisGit, 2021-11-13:
Hi @elminster 
new release [1.15.0](https://github.com/OllisGit/OctoPrint-PrintJobHistory/releases/tag/1.15.0) is available.
Please try the new version, maybe the issue is solved.

If not solved, please attach a screenshot from the error and also the current octoprint.log

Thx, in advance!
Olli

elminster, 2021-12-03:
I had switched off the check for dependencies. I have switched it back on and I am not getting the warning now, so it looks like it has been fixed. I have another printer with octoprint, I will test I get the warning on that one before I update the plugin just to double-check. But looks like the issue is fixed. Thanks

elminster, 2022-01-06:
Tested on both my printers and the issue is resolved. Thanks.

closed 2022-01-06 by elminster