OllisGit/OctoPrint-PrintJobHistory #159: [Enhancement] Force print job form before print

psi-chuffine, 2021-06-28:
It would be useful to be able to force users to complete the form before they can print a file.

OllisGit, 2021-11-13:
Hi @psi-chuffine!

hmmm... showing the edit job dialog before printing could be possible, but which fields should be mandatory? Or should it be just a reminder for e.g. add some notes?

psi-chuffine, 2021-11-15:
I'm using this one a shared printer at work to track which job to charge printer time to, so I've been having users fill out the notes field with that detail. 

A reminder should be enough, as the popup after print often gets ignored as users will just grab the print off the bed and skedaddle.

closed 2022-02-28 by github-actions[bot]

OllisGit, 2022-03-18:
Note to myself: Just a simple popup with the Note wysiwyg-textarea that (must) be filled. before printing.