OllisGit/OctoPrint-PrintJobHistory #160: [Enhancement] thumbnails background change

pejotigrek, 2021-07-09:
I use thumbnails generated from cura files - and while in OctoPrint they all have white background, when PrintJobHistory plugin uses them in job log - it makes the background black and that makes black prints almost invisible ;) how it can be changed?

thumbnails in OctoPrint's dialogues:
[image]
[image]

thumbnails in PrintJobHistory plugin:
[image]
[image]

OllisGit, 2021-11-13:
Hi @pejotigrek,

well...I convert the cura png into a jpg file and JPEG-Format doesn't have a transparent background.

Currently the printjob images are stored in the filesystem and the job-data is stored in a filebase-database.
Because I want to add the feature to use an external database (like postgres) I also need to change the image-storage situation.
So, when I am implementing the ext.db feature I will also change the png/jpg issue....but in the next release I will change it to a white background (hopefully there is no user with white-models ;-))

pejotigrek, 2021-11-13:
white models should be visible thanks to shadows :)
thanks!

OllisGit, 2021-11-13:
Hi @pejotigrek ,
new release [1.15.0](https://github.com/OllisGit/OctoPrint-PrintJobHistory/releases/tag/1.15.0) is available.

Please test and give me a feedback.

BR
Olli

pejotigrek, 2021-11-13:
> Please test and give me a feedback

will do! as soon as I'll get back to the work [next week], I've got the whole setup in the office

pejotigrek, 2021-11-15:
upgraded today, plugin manager shows: **PrintJobHistory (1.15.1)**.
tested with two quick prints [~50mins each]. no "add print job" dialogue window after successfully finished printing showed up, no job added to history log :(

after first print I double checked the settings - I had selected that only after successful print there should be a popup window; changed that to "after all prints", restarted system, tried another print - no change.

is it possible that this update broke something, or I should search for cause on my side? [FYI: before that, when I printed something last week, it worked fine, popup was showing after each print]

OllisGit, 2021-11-19:
Hi @pejotigrek,

please provide be the octoprint.log, maybe there is something logged that explains the issue.
[image]

pejotigrek, 2021-11-24:
> Hi @pejotigrek,
> 
> please provide be the octoprint.log, maybe there is something logged that explains the issue.

hi @OllisGit ,
forgive me the delay, got stuck with few projects at work and didn't managed even to read your answer - just saw the notification and snoozed it until today.

but! today before I send one of said projects to print, PrintJobHistory updated and... after printing all issues seems to be gone :)
first of all: "add print job" dialogue windows pops up, and after adding that to history - thumbnails are totally readable thanks to white background!

THANK YOU for the fix! I guess I can close the issue as it is resolved :)

closed 2021-11-24 by pejotigrek