OllisGit/OctoPrint-PrintJobHistory #161: Filament Usage In Statistics Popup Is Incorrect

donlook, 2021-07-09:
When you click on the "Statistic" link in the Print Job History tab, the filament usage statistics are overstated. I selected a date range that included only one job for this example.

[image]

[image]

When you click on the statistic report link, the filament usage numbers in this case are double what is shown in the job tab. The job tab reports the usage as 4.29 m and the statistic tab reports 8.58 m. The used weight is also doubled. 

[image]

I believe that this is due to the rows in the pjh_filamentmodel table for this job, in my case there are two, one for toolId "tool0" and one for toolID "total". Should the statistics report look only at the "total" tool ID row?

**pjh_printjobmodel table**
[image]

**pjh_filamentmodel table**
[image]

OllisGit, 2021-11-13:
Hi @donlook, 
new release [1.15.0](https://github.com/OllisGit/OctoPrint-PrintJobHistory/releases/tag/1.15.0) is available and implement  the patch from @hvraven (Thx, again)

Please test and give me a feedback.

BR
Olli

donlook, 2021-11-14:
Hello @OllisGit,
When I install this update, all my job history does not show. It looks like the database structure update is not completing successfully? Someone else posted an issue #174  with at trace that the pjh_costmodel table is missing.
I have been able to reinstall 1.14.0 and the history is visible again.
Don

donlook, 2021-11-14:
Hello @OllisGit,
The statistics are correct now.

PR #169 (2021-10-30): fix double counting bug in statistics overview
This fixes a bug in the statistic calculation where filament length,
weight and some other entries were doubled.
As already assumed by donlook in #161 this is due to the existence of
print job entries for every tool and a total entry. This simple fix
simply skips the total entries, only including the ones per tool.

Fixes #161
commit 8d5d3c1c (2021-10-30): fix double counting bug in statistics overview

This fixes a bug in the statistic calculation where filament length,
weight and some other entries were doubled.
As already assumed by donlook in #161 this is due to the existence of
print job entries for every tool and a total entry. This simple fix
simply skips the total entries, only including the ones per tool.

Fixes #161
closed 2021-11-14 by OllisGit