OllisGit/OctoPrint-PrintJobHistory #162: Search/Filter not working consistently

nigeljh99, 2021-07-12:
Really appreciate the plugin, it's is very very useful for finding previous settings and prints !
As part of that, i use the search quite a lot, but have found that sometimes it returns non results, even when typing inthe first 3 letters of a print title, yet osmetimes it pulls in results when i type a word in the middle .. haven't been able to find any consistent rhyme or reason as to why.
Not sure if it's a bug, or if i need to reset and reindex where the plugin stores these things but would welcome any guidance !

OllisGit, 2021-11-13:
Hi,
new release [1.15.0](https://github.com/OllisGit/OctoPrint-PrintJobHistory/releases/tag/1.15.0) is available and I changed the search behaviour.
If it is still not working, please attach your database and give my an example.

Please test and give me a feedback.

BR
Olli

nigeljh99, 2021-11-17:
Hi Olli,

Seems i have had version 1.15.1 installed for a while now on both printers, and did test and the search is now working and finding how i would expect, with things like partial matching and wildcard of words that are within a filename etc. So it's all looking good, and many thanks for listening and for this very useful plugin !

Kind Regards,
Nigel

closed 2021-11-19 by OllisGit