OllisGit/OctoPrint-PrintJobHistory #165: Issue: Cyclic deselecting "filament tracker" setting

SatarisGIT, 2021-08-02:
Hello!

I have problem with save settings for Print job history plugin. If I select "preferred filament tracker" as SpoolManager-Plugin, my setting constantly after some time will become deselected. After 3-4 prints or restart system, if I go to settings I can see thats selected preferred filament tracker is "None". When option is set as none, plugin totally not working correctly, not counting any weights or filament usages, so it is for me unusable. Is there any method for setting this ,,SpoolManager plugin" option permanently?

[image]

ivgh1992, 2021-08-05:
First of all, thank you @OllisGit for your great job. Sadly I`ve been having the same problem. Every time I shutdown or restart octoprint the selections jumps back to "None". Hope you can fix this. Thank you @OllisGit

eacxx, 2021-08-10:
Hi guys!
I'm in the same boat.
For me it's not after every restart or shutdown, but after 3 or 4 is happening for sure.
If somebody has a hint for those of us being less lucky then the rest, it would be more than welcome.

Thanks in advance.

donlook, 2021-08-11:
Look at issue #149

eacxx, 2021-08-12:
Thank you so much man!
And thanks to @ManuelMcLure especially.
I didn't tried yet, because I have a print rolling right now and I guess I can't restart Octoprint during it. I'm too new in this business so I don't know. I prefer to wait for the printer to finish.
But it looks to me that you really figured out the issue.
I'll come back with the feedback after I'll test.

OllisGit, 2021-11-13:
Duplicated with #149

closed 2021-11-13 by OllisGit

OllisGit, 2021-11-13:
Hi,
new release [1.15.0](https://github.com/OllisGit/OctoPrint-PrintJobHistory/releases/tag/1.15.0) is available and fix this issue.

BR
Olli