OllisGit/OctoPrint-PrintJobHistory #166: No record by Print Job History

J-Schuh, 2021-08-03:
Hello,

Print Job History cannot record any print history on my Prusa MK3S printer

Octoprint is latest version, Print job history 1.4

I also try to enter the record manually, I get same error.

Here is the log from Octoprint
[octoprint.log](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/6922203/octoprint.log)

Thank you in advance!

J-Schuh, 2021-08-03:
I tried to refresh install the plugin and disable and uninstall other plugins... does not help

OllisGit, 2021-11-13:
Hi @J-Schuh, 
I checked the log file and it looks like there is something wrong with the database.
Please download and attach the database to this issue.
[image]

jbfagotfede39, 2021-11-17:
Hello,

I have the same problem after update yesterday or today with last version.
Thanks
[printJobHistory.db.zip](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/7556915/printJobHistory.db.zip)

jbfagotfede39, 2021-11-18:
> Hello,
> 
> I have the same problem after update yesterday or today with last version. Thanks [printJobHistory.db.zip](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/7556915/printJobHistory.db.zip)

I don't now why, but it-s now OK. Sorry.

mikepaxton, 2021-11-18:
Having the same issue since last upgrade to Print Job History 1.15.1

None of my new jobs are being recorded in Print Job History. 
[Database+Log File.zip](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/7563659/Database%2BLog.File.zip)

OctoPrint 1.7.2
Filament manager 1.9.1
Cost Estimation 3.3.0

OllisGit, 2021-11-19:
Hi @mikepaxton,

there is something wrong with the filamentmanager-plugin:
```
octoprint.plugins.filamentmanager - ERROR - Failed to fetch profiles lastmodified timestamp: (psycopg2.OperationalError) server closed the connection unexpectedly
	This probably means the server terminated abnormally
	before or while processing the request.
 [SQL: 'SELECT modifications.changed_at \nFROM modifications \nWHERE modifications.table_name = %(table_name_1)s'] [parameters: {'table_name_1': 'profiles'}]

``` 
If your using the filebased database, please attach it to this issue. It is located in ```<op>/data/filamentmanager/filament.db```

OllisGit, 2021-11-19:
Hi @jbfagotfede39,
thanks for providing your database, but I also need your corresponding octoprint.log. Because the error could be totally different in your environment.

mikepaxton, 2021-11-19:
Hi @OllisGit 

Attached is the filament.db file you requested.
[filament.db.zip](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/7572703/filament.db.zip)

OllisGit, 2021-11-20:
Hi @mikepaxton,

sorry to say, but all tables (except the versioning) are empty -> no Spools/Profile data. 

I looked into your log file, but I could not figure out why all content is gone.
Some fragments from the log
```
2021-11-18 06:13:26,784 .... Updated remaining filament on spool 'Transparent - PETG (Overture)'
--> Working fine (last time)

2021-11-18 06:20:23,937 Command for core:reboot failed with return code -15:
--> Something was wrong with the reboot

2021-11-18 06:20:41,019 - octoprint.plugins.filamentmanager - ERROR - Failed to initialize database: (psycopg2.OperationalError) could not connect to server: Network is unreachable
	Is the server running on host "10.0.0.10" and accepting
	TCP/IP connections on port 5433?
--> No connection the database, but no the plugin tries to connect to an external database!?!?
```

Did you change the connection settings in the plugin-settings?
Did you change your ip-address of your server>
Is you sd-card old and "broken"?

mikepaxton, 2021-11-20:
Hi @OllisGit

So when I grabbed the filament.db file I totally forgot that I'm running the database on a server in docker.  So of course I would assume the database I sent you would be empty.   However, I'm not sure why the log says it couldn't connect to my server (10.0.0.10).  Well, on second thought every once in awhile Filament Manager is unable to connect which usually rebooting Octoprint solves it.  I'm guessing those couple of lines in the log reflected such an instance.

Yesterday evening I updated both Print Job History and Cost Estimation to the latest versions (1.15.2 & 3.4.0 respectively)and this seems to have fixed the issue.  The two prints that I've done since then have recorded all of the data in Print Job History.

madnlooney, 2021-11-22:
seems mine stopped recording. has been ok and then just stopped.

octoprint Version 1.7.2
latest printjobhistory 1.15.2
Cost Estimation (3.4.0)

[printJobHistory.zip](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/7582065/printJobHistory.zip)

[octoprint.zip](https://github.com/OllisGit/OctoPrint-PrintJobHistory/files/7582067/octoprint.zip)

J-Schuh, 2021-11-25:
> Hi @J-Schuh, I checked the log file and it looks like there is something wrong with the database. Please download and attach the database to this issue. [image]

Hi, 

My problem was automatically solved in mid October. Maybe because of upgrade of the Octoprint or the plugin. Which is not clear to me. During the troubleshooting, I also delete the database and make new ones, although this also did not help at that time. Thus, I have no old database to provide for debugging...

thanks anyway!

closed 2022-01-15 by OllisGit